signetlabdei/lorawan #18: Not able to complie

mamadur, 2018-04-01:
## Expected Behavior
Complitation should success

## Actual Behavior
Build fails

## Steps to Reproduce the Problem

  1. As mentioned, all the dependencies are installed and extracted the code 
  2. Build the code using "./test.py -s lorawan"

## Specifications

  - ns-3 version: Release 3-dev
  - lorawan module version:
  - Platform: Ubuntu 
  - Compiler:gcc

Please let me know if I missed anything or requires to update.

DvdMgr, 2018-04-03:
Hi!
Can you give me some further details on what is going wrong? What does the error message look like? Is it happening during configuration or compilation?

mamadur, 2018-04-04:
Thank you for your response.

Error is observed while compilation.

**Details**:
/usr/include/c++/4.8/bits/stl_list.h:1149:9: note:   template argument deduction/substitution failed:
../src/wifi/model/block-ack-manager.cc:927:51: note:   cannot convert ‘it’ (type ‘std::list<std::_List_iterator<ns3::BlockAckManager::Item> >::const_iterator {aka std::_List_const_iterator<std::_List_iterator<ns3::BlockAckManager::Item> >}’) to type ‘std::list<std::_List_iterator<ns3::BlockAckManager::Item> >::iterator {aka std::_List_iterator<std::_List_iterator<ns3::BlockAckManager::Item> >}’
               it = m_retryPackets.insert (it, item);
                                                   ^

Waf: Leaving directory `/home/mamadur/ns-3-dev/build'
Build failed
 -> task in 'ns3-wifi' failed (exit status 1): 
	{task 140185549375376: cxx block-ack-manager.cc -> block-ack-manager.cc.1.o}
['/usr/bin/g++', '-O0', '-ggdb', '-g3', '-Wall', '-Werror', '-std=c++11', '-Wno-error=deprecated-declarations', '-fstrict-aliasing', '-Wstrict-aliasing', '-fPIC', '-pthread', '-I.', '-I..', '-DNS3_BUILD_PROFILE_DEBUG', '-DNS3_ASSERT_ENABLE', '-DNS3_LOG_ENABLE', '-DHAVE_SYS_IOCTL_H=1', '-DHAVE_IF_NETS_H=1', '-DHAVE_NET_ETHERNET_H=1', '-DHAVE_PACKET_H=1', '-DHAVE_IF_TUN_H=1', '../src/wifi/model/block-ack-manager.cc', '-c', '-o',

DvdMgr, 2018-04-04:
The problem seems to be in the wifi module! It doesn't look like I can help you, since the problem is not with the lorawan module. 

I suggest you to visit the [ns-3-users google group](https://groups.google.com/forum/#!forum/ns-3-users) for support on this kind of issues!

closed 2018-04-04 by DvdMgr