signetlabdei/lorawan #20: ./waf build error

shikharoo7, 2018-04-24:
## Expected Behavior
lorawan modules should get configured

## Actual Behavior
Could not find a task generator for the name 'ns3-lorawan-master'
[image]

## Specifications

  - ns-3 version: 3.28
  - lorawan module version: 
  - Platform: Linux Mint 18.3
  - Compiler: Python 3.6.5, gcc 7.2.0

DvdMgr, 2018-04-24:
Hi!
Is the folder containing the module in the src directory named lorawan?

FENGFEINI, 2018-04-24:
I also had the same problem, how can I solve it?

FENGFEINI, 2018-04-25:
file:///home/ffn/Desktop/Screenshot%20-%2004252018%20-%2009:19:18%20AM.png

FENGFEINI, 2018-04-25:
The following figure is the directory
file:///home/ffn/Desktop/Screenshot%20-%2004252018%20-%2009:19:18%20AM.png

FENGFEINI, 2018-04-25:
/home/ffn/Downloads/Screenshot - 04252018 - 10:08:02 AM.png

FENGFEINI, 2018-04-27:
Sorry, I didn't upload screenshots. I still have a question of which version of ns-3 I need to install for using this  lora module.

DvdMgr, 2018-04-27:
ns-3.27 should be fine!

Anyway, make sure the name of the module folder in src is lorawan, we had a similar discussion on the repo’s gitter chat!

shikharoo7, 2018-04-27:
yup, that was the problem.
The folder was named as 'lorawan-master' and changing it to 'lorawan' solved the issue.
Thanks!

closed 2018-04-27 by shikharoo7