signetlabdei/lorawan #25: ./waf build error

brunoqt, 2018-09-18:
## Expected Behavior

lorawan modules should get configured
## Actual Behavior

Build failed
 -> task in 'ns3-lorawan' failed with exit status 1 (run with -v to display more information)
 -> task in 'ns3-lorawan' failed with exit status 1 (run with -v to display more information)
 -> task in 'ns3-lorawan' failed with exit status 1 (run with -v to display more information)
 -> task in 'ns3-lorawan' failed with exit status 1 (run with -v to display more information)

## Steps to Reproduce the Problem

  1.install ns-3, and clone this repository inside the src directory
  2../waf configure --enable-tests --enable-examples
  3../waf build

## Specifications

  - ns-3 version:3.28
  - lorawan module version:
  - Platform: Ubuntu 16.04
  - Compiler:2.7.12 Python

DvdMgr, 2018-09-19:
Hi!
Is the module folder named exactly "lorawan"?

brunoqt, 2018-09-19:
Yes. I've uploaded photos about the errors found.
[image]
[image]
[image]

DvdMgr, 2018-09-19:
Ok, which branch are you on? If you are not already on develop, try that one!

brunoqt, 2018-09-19:
I did not understand

afarhad, 2018-09-19:
Try to find the develop branch as shown in the picture below.
[cid:6ad4c90d-12f6-43b5-b7f8-6f9917117940]

brunoqt, 2018-09-19:
I did not receive the image.

afarhad, 2018-09-19:
Can you now?

brunoqt, 2018-09-19:
yes

DvdMgr, 2018-09-19:
To use the develop branch, you can use the following command, given from inside the lorawan folder:
git fetch && git checkout develop
Then, you can try building again

brunoqt, 2018-09-19:
Okay, I'll try and give you a return.

brunoqt, 2018-09-19:
It worked! Thanks!

closed 2018-09-19 by DvdMgr

saifmassoud, 2019-11-21:
hello everyone I ask for your help, I get many error when I build my project (using lorawan module) (using ns3.29 and ubuntu 18.04 LTS), the building error are linked below :)

seifeddine@seifeddine-Veriton-M480G:~/bake/source/ns-3.29$ sudo ./waf --run scratch/lorasimulation
[sudo] password for seifeddine:
Waf: Entering directory `/home/seifeddine/bake/source/ns-3.29/build'
[2793/2878] Compiling scratch/complete-lorawan-network-example.cc
[2794/2878] Compiling scratch/iotsimulation.cc
[2795/2878] Compiling scratch/simple-lorawan-network-example.cc
[2796/2878] Compiling scratch/lorasimulation.cc
../scratch/simple-lorawan-network-example.cc: In function ‘int main(int, char**)’:
../scratch/simple-lorawan-network-example.cc:66:7: error: ‘LoraChannel’ was not declared in this scope
Ptr<LoraChannel> channel = CreateObject<LoraChannel> (loss, delay);
^~~
../scratch/simple-lorawan-network-example.cc:66:7: note: suggested alternative:
In file included from ./ns3/lora-phy.h:30:0,
from ./ns3/end-device-lora-phy.h:32,
from ../scratch/simple-lorawan-network-example.cc:7:
./ns3/lora-channel.h:73:7: note: ‘ns3::lorawan::LoraChannel’
class LoraChannel : public Channel
^~~
../scratch/simple-lorawan-network-example.cc:66:18: error: template argument 1 is invalid
Ptr<LoraChannel> channel = CreateObject<LoraChannel> (loss, delay);
^
../scratch/simple-lorawan-network-example.cc:66:68: error: no matching function for call to ‘CreateObject<LoraChannel>(ns3::Ptr<ns3::LogDistancePropagationLossModel>&, ns3::Ptr<ns3::PropagationDelayModel>&)’
Ptr<LoraChannel> channel = CreateObject<LoraChannel> (loss, delay);
^
In file included from ./ns3/end-device-lora-phy.h:26:0,
from ../scratch/simple-lorawan-network-example.cc:7:
./ns3/object.h:528:8: note: candidate: template<class T> ns3::Ptr<T> ns3::CreateObject()
Ptr<T> CreateObject (void)
^~~~
./ns3/object.h:528:8: note: template argument deduction/substitution failed:
./ns3/object.h:540:8: note: candidate: template<class T, class T1> ns3::Ptr<T> ns3::CreateObject(T1)
Ptr<T> CreateObject (T1 a1)
^~~~
./ns3/object.h:540:8: note: template argument deduction/substitution failed:
./ns3/object.h:555:8: note: candidate: template<class T, class T1, class T2> ns3::Ptr<T> ns3::CreateObject(T1, T2)
Ptr<T> CreateObject (T1 a1, T2 a2)
^~~~
./ns3/object.h:555:8: note: template argument deduction/substitution failed:
./ns3/object.h:572:8: note: candidate: template<class T, class T1, class T2, class T3> ns3::Ptr<T> ns3::CreateObject(T1, T2, T3)
Ptr<T> CreateObject (T1 a1, T2 a2, T3 a3)
^~~~
./ns3/object.h:572:8: note: template argument deduction/substitution failed:
./ns3/object.h:591:8: note: candidate: template<class T, class T1, class T2, class T3, class T4> ns3::Ptr<T> ns3::CreateObject(T1, T2, T3, T4)
Ptr<T> CreateObject (T1 a1, T2 a2, T3 a3, T4 a4)
^~~~
./ns3/object.h:591:8: note: template argument deduction/substitution failed:
./ns3/object.h:612:8: note: candidate: template<class T, class T1, class T2, class T3, class T4, class T5> ns3::Ptr<T> ns3::CreateObject(T1, T2, T3, T4, T5)
Ptr<T> CreateObject (T1 a1, T2 a2, T3 a3, T4 a4, T5 a5)
^~~~
./ns3/object.h:612:8: note: template argument deduction/substitution failed:
./ns3/object.h:635:8: note: candidate: template<class T, class T1, class T2, class T3, class T4, class T5, class T6> ns3::Ptr<T> ns3::CreateObject(T1, T2, T3, T4, T5, T6)
Ptr<T> CreateObject (T1 a1, T2 a2, T3 a3, T4 a4, T5 a5, T6 a6)
^~~~
./ns3/object.h:635:8: note: template argument deduction/substitution failed:
./ns3/object.h:660:8: note: candidate: template<class T, class T1, class T2, class T3, class T4, class T5, class T6, class T7> ns3::Ptr<T> ns3::CreateObject(T1, T2, T3, T4, T5, T6, T7)
Ptr<T> CreateObject (T1 a1, T2 a2, T3 a3, T4 a4, T5 a5, T6 a6, T7 a7)
^~~~
./ns3/obj

DvdMgr, 2019-11-22:
Hi, it looks like this error is because of some error in your own lorasimulation.cc script - there's nothing we can do about it!

saifmassoud, 2019-11-22:
hello everyone I ask for your help, I get many error when I build my project (using lorawan module) (using ns3.29 and ubuntu 18.04 LTS), the building error are linked below :)
[image]
[image]
[image]

DvdMgr, 2019-11-22:
Again, the problem stems from a script in your scratch folder. `./waf --run complete-network-example` works fine on my machine!

saifmassoud, 2019-11-22:
> Again, the problem stems from a script in your scratch folder. `./waf --run complete-network-example` works fine on my machine!

Thanks @DvdMgr for your reply, could you tel me which version of ns-3 with which lorawan module can I use? if you have a full tutorial for installing, could you send it to me?

DvdMgr, 2019-11-22:
The module is compatible with ns-3.30 - for installation, you can follow
the instructions in the readme!

saifmassoud, 2019-11-22:
Thanks @DvdMgr i'll verfiy, I hope that work :p