signetlabdei/lorawan #26: Example program not found

rizkyarlin, 2019-01-24:
## Expected Behavior
Running example program

## Actual Behavior
Error message: 
program 'complete-lorawan-network-example' not found; available programs are: ....
All three examples are not found

## Steps to Reproduce the Problem

  1. Following all the **Getting Started** on [https://github.com/signetlabdei/lorawan](https://github.com/signetlabdei/lorawan)
  2. No configure error
  3. No build error
  4. Passed test
  5. Executing ./waf --run program-name

## Specifications

  - ns-3 version: 3.29
  - lorawan module version: 0.2.0
  - Platform: macOS Mojave
  - Compiler: cc 10.0.0

DvdMgr, 2019-01-24:
Hi,
can you try running ./waf --run complete-network-example?

I shortened the names of the examples in a recent commit, and I probably forgot to update the readme. In any case, you can always find out which examples are available in the module by going inside the examples folder!

DvdMgr, 2019-03-11:
Closing due to inactivity

closed 2019-03-11 by DvdMgr