signetlabdei/lorawan #28: Can't use EnablePacketTracking(filename) to tracking

qtrunguet, 2019-03-08:
## Expected Behavior
Performance file in complete-network-example.cc have data

## Actual Behavior
That file is empty

## Steps to Reproduce the Problem

  1.run complete-network-example.cc
  2. Open performance file
  3. This is empty

## Specifications

  - ns-3 version: 3.28
  - lorawan module version:  v0.2.0
  - Platform: ubuntu
  - Compiler: cc

DvdMgr, 2019-03-11:
Hi, the argument to the EnablePacketTracking method is a leftover of a previous implementation. In the current one, performance is printed directly on the standard output. I'll clean this up in the near future!

qtrunguet, 2019-03-12:
At example: complete-network-example.cc the result printed but i think something has wrong because i got it:
`Completed configuration
Running simulation...
Computing performance metrics...
Successful with 1 | Successful with 2 | Successful with 3 | Successful with 4 | Successful with 5 | Successful with 6 | Successful with 7 | Successful with 8 | Failed after 1 | Failed after 2 | Failed after 3 | Failed after 4 | Failed after 5 | Failed after 6 | Failed after 7 | Failed after 8 | Average Delay | Average ACK Delay | Total Retransmission amounts || PHY Total | PHY Successful | PHY Interfered | PHY No More Receivers | PHY Under Sensitivity | PHY Lost Because TX
0 0 0 0 0 0 0 0  | 0 0 0 0 0 0 0 0  | 0 0  | 0 || 200 193 7 0 0 0 `

DvdMgr, 2019-03-12:
The metrics look good to me: since by default the EDs don't require confirmation, they only transmit the packets once and you get the PHY layer performance.

You can safely ignore the zeros in the first part, since they are only filled when confirmed transmissions are used.

closed 2019-03-14 by DvdMgr