signetlabdei/lorawan #33: SNIR Calculation

Prapi123, 2019-08-30:
## Expected Behavior
Hello
1. I want to calculate SNIR and SNR in the network. I am making some modifications to the complete-network-example. I can see in the results that SNIR is always inf db as interference power is 0. But on the contrary the Result matrix shows PHY interfered = 6. Please help me understand this phenomenon.

2. I am using the below code to get SF of 1st receive window

Ptr<LoraMac> edMac1 = endDevices.Get (0)->GetDevice(0)->GetObject<LoraNetDevice> ()->GetMac ();
Ptr<EndDeviceLoraMac> edLoraMac1 = edMac1->GetObject<EndDeviceLoraMac> ();
LoraDeviceAddress Address = edLoraMac1->GetDeviceAddress();
Ptr<EndDeviceStatus> edStatus = CreateObject<EndDeviceStatus> (Address, edLoraMac1);
edStatus->SetFirstReceiveWindowSpreadingFactor (9);
 NS_LOG_INFO ("GetFirstReceiveWindowSpreadingFactor: " << edMac1->GetFirstReceiveWindowSpreadingFactor());

The SF is not displayed. Also the function GetSfFromDataRate() cannot display SF for my code.

3. In the example : Complete-network-example, performance and building files are empty.  Nothing is being recorded in these files.

4.  If I change number of Gateways in complete-network-example to 3 and Enddevices  to 200.
Total Phy number is 600. I expected it to be 200 only. Please help me understand this as well

## Steps to Reproduce the Problem

  1.  run complete-network-example.cc
 

## Specifications

  - ns-3 version: 3.28
  - lorawan module version: v0.2.0
  - Platform: ubuntu
  - Compiler: cc

DvdMgr, 2019-09-03:
Hi!
In the future please open multiple issues, one for each problem you encounter. This will make it easier to follow the discussion. For now, let's try and address all 4 issues here.

1. An execution of complete-network-example gives 7 interfered packets. If I uncomment the LoraInterferenceHelper log component and run `./waf --run complete-network-example 2>&1 | grep SNIR`, I get multiple instances in which the SNIR is negative, and could thus lead to packet losses. In other words, the SNIR is not always inf. Am I missing something?
2. Besides the fact that the code does not compile, I don't get what you are doing here. You can simply use the GetFirstReceiveWindowDataRate () method on any instance of EndDeviceLoraMac to get the DataRate that will be used in the first receive window by that device. 
3. The buildings file is empty because the `realisticChannelModel` variable is set to false by default, and thus no buildings are created. If you set it to true, the buildings.txt file will be filled. performance.txt is empty as of now because we are moving to a better packet tracker, which is implemented in the adr branch for now. In the meanwhile, all performance is printed directly to the standard output.
4. Since you have three gateways, each PHY packet generated by the devices will be received three times, by all the three gateways. Since each device generates exactly one packet, since that packet is received (not necessarily correctly) by three gateways, and since you have 200 devices, you get 600 PHY packets in your simulation.

Prapi123, 2019-09-03:
Hello,

Thanks for reply and helping out in the issue.
For 2. Point I am attaching a png file of my result. where i cannot display my spreadingfactor and offset.
Please have a look.

[image]

DvdMgr, 2019-09-04:
Oh, this is a known issue tied to the fact that both the Data Rate and Spreading Factor values are saved as uint8_t internally. If you wrap the value returned by GetFirstReceiveWindowDataRate() inside an unsigned() it should work out!

Prapi123, 2019-09-04:
Thanks for the reply. It worked

closed 2019-09-04 by DvdMgr