signetlabdei/lorawan #34: Packet error rate

Prapi123, 2019-09-03:
Hello, 
Is there already a class defined to calculate packet error rate?
I am hoping to calculate packet error rate at each gateway. 
Is it possible to get number of packet received and number of packet lost at each gateway  in case of multiple gateways?

nmatni, 2019-09-03:
I don't think that already have such a  thing. But you could add a counter in form of a vector, for example, to identify each gw.

Prapi123, 2019-09-04:
Thanks for your Reply. I will try do it in this way.

DvdMgr, 2019-09-05:
> Is it possible to get number of packet received and number of packet lost at each gateway in case of multiple gateways?

Yes, it is! If you are interested in GW-level statistics, we have methods to compute those in the adr branch, both at the MAC and at the PHY level. Check out the `PrintPhyPacketsPerGw` method in the `lora-packet-tracker.h` and `lora-packet-tracker.cc` files on the adr branch! The `adr-example.cc` file on that branch, towards the end, has an example of how you can get the packet tracker that was used during simulation and that can be used to print various kinds of statistics.

Prapi123, 2019-09-05:
Hello ,

Thank you for the response.
I would like to confirm about one parameter in the method  PrintPhyPacketsPerGw (Time startTime, Time stopTime, **int systemId**) .
Is systemId here the address of the gateway?
What is the best way to return system id ?

kaustubhshrotri, 2019-09-09:
Hello,

I would like to know if there is a method to track number of packets sent and received from each enddevices to

DvdMgr, 2019-09-09:
Currently this is not implemented, but shouldn't be too hard to do. Right now I don't have much time I can allocate to this, but I can definitely guide you if you want to tackle this!

In case you want to look into this, all packets are tracked in the `lora-packet-tracker.cc/.h` files. I recommend using the `adr` branch, as it features the latest version of the packet tracker.

kaustubhshrotri, 2019-09-09:
Hi,
Thanks for answering.
I am working on calculation of packet error rate at end device level using this module.
Should I open of new issue or I can post my quetions in this thread?

DvdMgr, 2019-09-09:
Best to open another issue to track further developments, so that I can close this one!

closed 2019-09-09 by DvdMgr

Prapi123, 2019-09-17:
**Small Information for other users who will be reading this thread in future:**
You can access id with GetId() method on desired nodes.
For me SystemId() would always give 0 for every node.
GetId() worked for me