signetlabdei/lorawan #35: Add multiple types of devices

nmatni, 2019-09-03:
First i wonder how could I add a high priority and a low priority devices, changing few thing as BW, SF and CR

DvdMgr, 2019-09-04:
CR and BW are not modifiable as of now. The SF can be changed on individual devices by getting their EndDeviceLoraMac and calling the SetDataRate method, which allows you to set the Data Rate, which is essentially the reverse of the SF - SF12 = DR0, ... SF7 = DR5.

However, I'm not sure how changing this value would implement a concept of high and low priority devices!

nmatni, 2019-09-04:
Actually, the idea is change SF, BW and CR  to use for different types of application

DvdMgr, 2019-09-05:
I see, unfortunately, our current channel model only supports the default code rate and a 125 kHz bandwidth. If you want, we can talk about extending it to support other values, too!

DvdMgr, 2019-11-05:
Closing due to inactivity.

closed 2019-11-05 by DvdMgr