signetlabdei/lorawan #36: Packet Error rate at End Device level.

kaustubhshrotri, 2019-09-09:
Hello Group,

I am new to this LoRaWAN simulator tool/Module. Please guide me on how can I approach to create a method for calculation of packet error rate at end device level.

Prapi123, 2019-09-09:
I am working on the similar topic. 
On the adr branch you can look for lora-helper files. There is a method called GetPacketTracker(), which returns pointer to m_packettracker. The packet tracker also has sender id. which can help to identify sending end nodes. I started off with this.

kaustubhshrotri, 2019-09-17:
Hello,

My code is below to access m_packetracker

LoraHelper helper = LoraHelper ();
LoraPacketTracker& packettracker = helper.GetPacketTracker();  
 for (auto itPhy = packettracker.begin ();
       itPhy != packettracker.end ();
       ++itPhy)
  {
    NS_LOG_INFO("Packet: " << itPhy);
  }

I want to access the packet in m_packetracker to get information of the packet.

I receive the error shown in picture

[image]

DvdMgr, 2019-09-18:
It looks like you are trying to access the packet tracker as it were a vector: instead, you should treat it as an object, which exposes the methods defined in the `lora-packet-tracker.h` file!

Prapi123, 2019-10-02:
Hello Both,

Below is my draft code for a method I defined in lora-packet-tracker.h/.cc. 
The method is still in progress, using The method I want to calculate the packet error rate at end devices.

  void LoraPacketTracker::EndDevicesPER (NodeContainer endDevices, Time startTime, Time stopTime)
  {
    NS_LOG_FUNCTION(this << startTime << stopTime);
    for (NodeContainer::Iterator j = endDevices.Begin (); j != endDevices.End (); ++j)
    {
      double sent = 0;
      double received = 0;
      Ptr<Node> object = *j;
      Ptr<MobilityModel> position = object->GetObject<MobilityModel> ();
      Vector pos = position->GetPosition ();
      uint32_t senderID = object->GetId ();
      //std::cout<< " Position: " << pos.x << "  " << pos.y << std::endl;
      //std::cout<< " id: " << senderID << std::endl;
      for(auto it = m_macPacketTracker.begin (); it != m_macPacketTracker.end (); ++it)
      {
        uint32_t edid = (*it).second.senderId;
        std::cout<< "edid: " << edid << std::endl;
        if((*it).second.sendTime >= startTime && (*it).second.sendTime <= stopTime)
        {
          if(senderID == edid)
          {
            sent++;
            if ((*it).second.receptionTimes.size ())
            {
              received++;
            }
          }
        }
      
      }
      //std::cout<< " sent: " << sent << std::endl;
      //std::cout<< " received: " << received << std::endl;
      double PER = (sent - received)/sent;
      if (PER > 0.5)
      {
        std::cout<< " Might need one more gateway near" << pos.x << " " << pos.y << " as the PER at node" << senderID << "is " << PER << std::endl;
      //std::cout<< " PER: " << PER << std::endl;
      }
    }

  }

In the above code I cannot get **edid** from the m_packetTracker. 
I tried to print edid but no success. 
As you can see from the code, i want to identify packets with the senderId.

Please let me know, what I am doing wrong and why is the error.

Thanks

@DvdMgr @kaustubhshrotri

Prapi123, 2019-10-03:
The above error is solved

Prapi123, 2019-10-08:
void LoraPacketTracker::EndDevicesPER (NodeContainer gateways, NodeContainer endDevices, Time startTime, Time stopTime, std::string filename)
  {
    const char * c = filename.c_str ();
    std::ofstream outputFile;
    if (Simulator::Now () == Seconds (0))
    {
      // Delete contents of the file as it is opened
      outputFile.open (c, std::ofstream::out | std::ofstream::trunc);
    }
    else
    {
      // Only append to the file
      outputFile.open (c, std::ofstream::out | std::ofstream::app);
    }

    
    for (NodeContainer::Iterator j = endDevices.Begin (); j != endDevices.End (); ++j)
    { 
      double sent = 0;
      double received = 0;
      Ptr<Node> object = *j;
      Ptr<MobilityModel> position = object->GetObject<MobilityModel> ();
      Vector pos = position->GetPosition ();
      uint32_t senderID = object->GetId ();
      for(auto it = m_packetTracker.begin (); it != m_packetTracker.end (); ++it)
      {
        uint32_t edid = (*it).second.senderId;
        if((*it).second.sendTime >= startTime && (*it).second.sendTime <= stopTime)
        {
          if(senderID == edid)
          {           
            sent++;
            for (NodeContainer::Iterator j = gateways.Begin (); j != gateways.End (); ++j)
            {
              Ptr<Node> object = *j;
              uint32_t gwID = object->GetId ();
              if ((*it).second.outcomes.count(gwID) > 0)
              { 
                switch ((*it).second.outcomes.at (gwID))
                {
                  case RECEIVED:
                  {
                    received++;
                    break;
                    
                  }
                  
                  default:
                  {
                    break;
                    
                  }
                  
                }
                 
              }
            
            }
    
                 
          }
        }
      
      }
      double PER = (sent - received)/sent;
      outputFile << Simulator::Now ().GetSeconds () << " PER at node " << senderID << " is " << PER << std::endl;
      if (PER > 0.4)
      {
        outputFile << " PER at node " << senderID << " is " << PER << " Might need one more gateway near " << pos.x << " " << pos.y << std::endl;
        
      }
    }
    outputFile.close ();
  }

Hello Both,
In the above I am trying to calculate PER at each end device.
My concern is the file is recording only data at Time = 0 seconds.
I am not getting why is this error. 

The output is like below , the displays time, EndDevice id, and PER
0 0 0.666667
0 1 0
0 2 0
0 3 0
0 4 0
0 5 0
0 6 0
0 7 0
0 8 0
0 9 0.333333
0 10 0.333333
0 11 0
0 12 0

Thanks
@DvdMgr @kaustubhshrotri

kaustubhshrotri, 2019-10-09:
This Issue has been resolved. You can close the issue.

Prapi123, 2019-10-09:
The issue is resolved

closed 2019-10-09 by DvdMgr