signetlabdei/lorawan #37: ./waf build error

ypwang36, 2019-09-15:
## Expected Behavior
lorawan modules should get configured

## Actual Behavior
In file included from ./ns3/lora-phy.h:30:0,
                 from ./ns3/end-device-lora-phy.h:32,
                 from ../src/lorawan/examples/simple-network-example.cc:6:
./ns3/lora-channel.h:84:23: error: conflicting return type specified for ‘virtual std::size_t ns3::lorawan::LoraChannel::GetNDevices() const’
   virtual std::size_t GetNDevices (void) const;
                       ^
In file included from ./ns3/lora-channel.h:30:0,
                 from ./ns3/lora-phy.h:30,
                 from ./ns3/end-device-lora-phy.h:32,
                 from ../src/lorawan/examples/simple-network-example.cc:6:
./ns3/channel.h:68:20: error:   overriding ‘virtual uint32_t ns3::Channel::GetNDevices() const’
   virtual uint32_t GetNDevices (void) const = 0;
                    ^
In file included from ./ns3/end-device-lora-phy.h:26:0,
                 from ../src/lorawan/examples/simple-network-example.cc:6:
./ns3/object.h: In instantiation of ‘ns3::Ptr<T> ns3::CreateObject(T1, T2) [with T = ns3::lorawan::LoraChannel; T1 = ns3::Ptr<ns3::LogDistancePropagationLossModel>; T2 = ns3::Ptr<ns3::PropagationDelayModel>]’:
../src/lorawan/examples/simple-network-example.cc:66:68:   required from here
./ns3/object.h:557:28: error: invalid new-expression of abstract class type ‘ns3::lorawan::LoraChannel’
   return CompleteConstruct (new T (a1,a2));
                            ^
In file included from ./ns3/lora-phy.h:30:0,
                 from ./ns3/end-device-lora-phy.h:32,
                 from ../src/lorawan/examples/simple-network-example.cc:6:
./ns3/lora-channel.h:73:7: note:   because the following virtual functions are pure within ‘ns3::lorawan::LoraChannel’:
 class LoraChannel : public Channel
       ^
In file included from ./ns3/lora-channel.h:30:0,
                 from ./ns3/lora-phy.h:30,
                 from ./ns3/end-device-lora-phy.h:32,
                 from ../src/lorawan/examples/simple-network-example.cc:6:
./ns3/channel.h:75:26: note: 	virtual ns3::Ptr<ns3::NetDevice> ns3::Channel::GetDevice(uint32_t) const
   virtual Ptr<NetDevice> GetDevice (uint32_t i) const = 0;

## Steps to Reproduce the Problem

  1. clone this repository inside the src directory
  2../waf configure --enable-tests --enable-examples
  3../waf build
[image]

## Specifications

  - ns-3 version: 3.28
  - lorawan module version:  branch develop
  - Platform:Ubuntu 16.04
  - Compiler:2.7.12 Python

ypwang36, 2019-09-16:
when I use branch gh-pages, I can success. But the gh-pages version is too old, and I want to use the newer version

DvdMgr, 2019-09-16:
Can you try compiling on ns-3.29 or ns-3.30?

ypwang36, 2019-09-22:
thanks！ I compiled successful in ns-3.30!

closed 2019-09-22 by ypwang36