signetlabdei/lorawan #38: EndDeviceStatus::InsertReceivedPacket()

Prapi123, 2019-09-17:
Hello All,

I have a question to ask for method InsertReceivedPacket() in EndDeviceStatus files. 
The method shows the notification "Packet Received for the first time"  but the total number of packets received is shown Zero.
I feel It should at least show Total number of packets received as 1.
Below is the image.
[image]

DvdMgr, 2019-09-18:
Hi, can you provide any instructions to reproduce this?

Prapi123, 2019-09-19:
Hello,
I have added the header file **#include "ns3/end-device-status.h"**  and code line **LogComponentEnable ("EndDeviceStatus", LOG_LEVEL_ALL);**  in the example complete-network-example.cc file.

**A small code before Computing matrix, given below.**
for (NodeContainer::Iterator j = endDevices.Begin ();
       j != endDevices.End (); ++j)
    {
      Ptr<Node> node = *j;
      Ptr<LoraNetDevice> loraNetDevice1 = node->GetDevice (0)->GetObject<LoraNetDevice> ();
      Ptr<LoraPhy> phy = loraNetDevice1->GetPhy ();

      Ptr<LoraMac> mac = loraNetDevice1->GetMac ();
      Ptr<EndDeviceLoraMac> edLoraMac = mac->GetObject<EndDeviceLoraMac> ();
      //edLoraMac->OnRxParamSetupReq ( 0, 3, 869.525);

      LoraDeviceAddress Address = edLoraMac->GetDeviceAddress();
      Ptr<EndDeviceStatus> edStatus = CreateObject<EndDeviceStatus> (Address, edLoraMac);
      EndDeviceStatus::ReceivedPacketList list = edStatus->GetReceivedPacketList();
     
    } 

Run the complete-network-example.cc file and and you can get these results