signetlabdei/lorawan #39: Multihop

peggyopal, 2019-09-26:
Not a defect/problem/strange behaviour per se, but I'd like to contribute multi-hop capability to the module.

DvdMgr, 2019-09-27:
Hi, what do you mean exactly with multi-hop? A mesh network between End Devices, in which each ED can act as a forwarder of messages? Is there any reference protocol you are going to follow to implement this kind of network architecture?

peggyopal, 2019-09-27:
Yes, a mesh network between EDs where each can act as a forwarder of messages. There are a couple of papers that I'll reference. Essentially I'd like to contribute something that would allow others to use the LoRaWAN module to create a proof of concept for protocol ideas they'd like to implement in the real world but can't test the scalability due to cost of devices.

peggyopal, 2019-09-27:
Yes, a mesh network between EDs where each can act as a forwarder of messages. There are a couple of papers that I'll reference. For more details I'd prefer to email if that is okay.

DvdMgr, 2019-09-30:
Ok, let's move this to email. I'll still keep the issue open to publicly track developments.

JovanRamjewan, 2020-01-15:
Hey, was the multi-hop capability added? If not, may I offer to help in adding the functionality?

FireHunter27, 2020-03-06:
Hey, can we use multi-hop capability?

DvdMgr, 2020-03-06:
Since there are multiple people that seem to be interested in this, I think you guys should coordinate - feel free to use this issue to do that! To get things started, for instance, I think you could describe more in detail what you want to achieve/simulate.

If needed, I can give advice and help a little bit with the design and integration of Multihop functionality.

JovanRamjewan, 2020-03-09:
Gladly! The goal for me is a LoRa mesh or ad-hoc network. Basically, only a couple of end devices can communicate with the GW, and all of the other end devices acts as a forwarder to the GW.  The quickest example I can find is https://nootropicdesign.com/projectlab/2018/10/20/lora-mesh-networking/

DvdMgr, 2020-03-20:
I think this is a good reference for what we want to achieve. Do you have a more specific design reference that contains the format of the messages that are exchanged between nodes, for instance, or the protocol the nodes follow to decide where to forward? It seems like they rely heavily on RadioHead to get the mesh network working, so we would probably need to replicate part of that codebase. I think there might be some papers in the literature showing proposals for mesh networking in LoRa, unless you want to design some original scheme I think we could use one of those as reference.

peggyopal, 2020-03-30:
Even if you want to design some original scheme, I'd still recommend replicating a previously research and tested protocol to make sure that you can achieve similar results before implementing an original design.  @JovanRamjewan

peggyopal, 2020-03-30:
I'm finally starting to work towards this.  I think I have finished the implementation for Class C devices, but I have someone looking at it before I create a PR to introduce it to the repo.  
SO!  @JovanRamjewan have you started implementation yet?  Where would you like to communicate about this?  Here?  Email?

I have a few ideas about implementation, one of them being I'm starting to create a helper class that will set the mac type for the protocol, that way it can be a simple setting to add to the code.  I can and will explain more, especially in the PR when it's finished.  But adding this class should make it easy for others to add more protocols in the future as well.

fedepaj, 2020-04-11:
@peggyopal could you share the papers you are working with?

houssembsir, 2020-05-12:
hi all , i was talking with David about implementing the class c and he mentioned that you (@peggyopal ) are working on it . 
it will be a great help if you can share your implementation.  
also im working on application layer for the network server . the goal is to have downlink data from network server not only ACK to evaluate the lorawan network performance on downlink.

peggyopal, 2020-05-12:
Hi @houssembsir yeah, I actually finished the implementation for Class C over a month ago.  
Can be seen [here](https://github.com/peggyopal/lorawan/pull/7).
I need to do a bit of refactoring and edit the documentation a bit but it works as it stands right now.

peggyopal, 2020-05-12:
Sorry @fedefava86 I did write up which papers I was working at, but apparently it didn't send. 
I'm currently implementing this [paper](https://ieeexplore.ieee.org/document/8436722/) but was also looking at others, such as [this](https://ieeexplore.ieee.org/document/8048465) paper.   If you like I can point you in the direction of other papers I've read as well.

houssembsir, 2020-05-12:
hi @peggyopal  , thank you for sharing . i will have a look at your implementation.

fmc5046, 2021-02-17:
Hi all, I'm curious how this implementation is going I'm interested in this functionality as well and would be happy to help out if needed

SalahEddineElgharbi, 2023-01-05:
Hi all,  I'm Interested (@DvdMgr @peggyopal @fmc5046 @JovanRamjewan @fedepaj )  

could anyone please interact with me about this subject, am still also work to enhance Multihop communication in lorawan as the routing part !