signetlabdei/lorawan #40: Identify ED in downlink packets

Fundoprajakta, 2019-10-07:
Hello,

First I was wondering how to identify recipient node in each downlink packet. 

And second how to identify reply packet and request packet pair.

Thanks

DvdMgr, 2019-10-07:
Hi!
To get the address of the recipient ED from a DL packet, you can follow these steps:
- Make a copy of the packet (packets are usually `const`, so you can only work on copies)
- De-serialize the Mac and Frame headers
- Get the Address field of the Frame header

In short, this can be accomplished with this code:
```c++
  LoraMacHeader mHdr;
  LoraFrameHeader fHdr;
  fHdr.SetAsDownlink ();
  Ptr<Packet> myPacket = packet->Copy ();
  myPacket->RemoveHeader (mHdr);
  myPacket->RemoveHeader (fHdr);
  LoraDeviceAddress address = fHdr.GetAddress ();
```

As for identifying reply packet and request packet pairs, you can try matching sender addresses with the address you find with the code above.

Fundoprajakta, 2019-10-07:
Hello !

Thanks for replying to the query. I will try this and respond to you.

I would like to know one more.
The packet tracker m_packettracker defined in lora-packetracker.cc/h contains all packets? Both downlink packets and uplink packets?

DvdMgr, 2019-10-07:
The packet tracker only saves the uplink packets. The `LoraPacketTracker::m_reTransmissionTracker` structure, however, contains a flag indicating success of MAC packets, based on whether an ACK was correctly received in at least one of the transmission attempts.

Fundoprajakta, 2019-10-07:
The idea is to get ratio of number of downlink packets sent by network server to number of packets received by end devices. 
m_reTramsmissionTracker can be helpful in this goal?

DvdMgr, 2019-10-07:
If you want PHY level statistics, I think you are better off connecting to the trace sources, and keeping the count yourself. For instance, you can connect to the `StartSending` trace source at the GWs, and to the `ReceivedPacket` trace source at the EDs. Both trace sources are provided by the LoraPhy class, which is the base for both SimpleGatewayLoraPhy and SimpleEndDeviceLoraPhy. Once you hook those trace sources to your functions, you can update your own counters and get the ratio you need at the end of the simulation.

Fundoprajakta, 2019-10-10:
Hello,

I tried to record downlink packet in a PhyPacketData similar to m_packetTracker for downlink packets.
with below code in lora-packet-tracker.cc

void
LoraPacketTracker::TransmissionCallbackDownLink (Ptr<Packet const> packet, uint32_t gwId)
{

  Ptr<Packet> packetCopy = packet->Copy ();

  // Remove the Mac Header to get some information
  LoraMacHeader mHdr;
  packetCopy->RemoveHeader (mHdr);
  if (!mHdr.IsUplink())
    {
      NS_LOG_INFO ("PHY packet " << packet
                                 << " was transmitted by gateway "
                                 << gwId);
      // Create a packetStatus
      PacketStatusDownlink status;
      status.packet = packet;
      status.sendTimeDownlink = Simulator::Now ();
      status.senderIdDownlink = gwId;

      m_packetTrackerDownlink.insert (std::pair<Ptr<Packet const>, PacketStatusDownlink> (packet, status));
    }
}

**Some how I cannot record downlink packets. the size of m_packetTrackerDownLink is always 0. Any help will be appreciated.**

P.S. : In Lora-helper.cc , I added the trace source StartSending in for gateway and the callback is directed to TransmissionCallbackDownLink.

Thanks and Regards

Fundoprajakta, 2019-10-11:
Hello,

I tried to create a PhyPacketData similar to m_packetTracker to record downlink packet.
Unfortunately it is not recording any packets.
below is the short code. Please let me know where I am making mistake.

void
LoraPacketTracker::TransmissionCallbackDownlink (Ptr<Packet const> packet, uint32_t gwId)
{
  if (!IsUplink(packet))
    {
      NS_LOG_INFO ("PHY packet " << packet
                                 << " was transmitted by gateway "
                                 << gwId);
      // Create a packetStatus
      PacketStatusDownlink status;
      status.packetDownlink = packet;
      status.sendTimeDownlink = Simulator::Now ();
      status.senderIdDownlink = gwId;

      m_packetTrackerDownlink.insert (std::pair<Ptr<Packet const>, PacketStatusDownlink> (packet, status));
    }
}

P.S : I also connected StartSending Trace source in lora-helper.cc which directs to call back TransmissionCallbackDownlink().

Thanks

Fundoprajakta, 2019-10-11:
Hello,
I tried to create a PhyPacketData similar to m_packetTracker to record downlink packet.
Unfortunately it is not recording any packets.
below is the short code. Please let me know where I am making mistake.

`void
LoraPacketTracker::TransmissionCallbackDownlink (Ptr<Packet const> packet, uint32_t gwId)
{
  Ptr<Packet> packetCopy = packet->Copy ();
  LoraMacHeader mHdr;
  packetCopy->RemoveHeader (mHdr);
  if (!mHdr.IsUplink ())
    {
      NS_LOG_INFO ("PHY packet " << packet
                                 << " was transmitted by gateway "
                                 << gwId);
      // Create a packetStatus
      PacketStatusDownlink status;
      status.packetDownlink = packet;
      status.Downlink= Simulator::Now ();
      status.senderIdDownlink = gwId;

      m_packetTrackerDownlink.insert (std::pair<Ptr<Packet const>, PacketStatusDownlink> (packet, status));
    }
}`

**P.S : I also connected StartSending Trace source in lora-helper.cc which directs to call back TransmissionCallbackDownlink().**

Thanks

DvdMgr, 2019-10-14:
Hi! Can you fork the lorawan repository and upload these changes to a branch on your own fork? This would make it significantly easier for me to check out your changes!

Fundoprajakta, 2019-10-14:
Hello!
Sure! I will do this now. 
Also I created a method to calculate PER for each node. 
I would appreciate, if you have a look on it and provide feedback.
It works fine for my purpose.

Fundoprajakta, 2019-10-14:
Hello!
I have forked the lorawan repository and uploaded the changes.
Please check the **adr branch : lora-helper.cc/h and lora-packet-tracker.cc/h file** and let me know your feedback.

Thanks!

DvdMgr, 2019-10-29:
I'm sorry for the delay - I now have some time to check this out, is this still an issue?

Fundoprajakta, 2019-10-29:
Hello Thanks for responding. Yes I am still working to capture downlink packets

DvdMgr, 2019-10-29:
I checked out the code on your adr branch and it doesn't seem to compile - can you confirm this?

Fundoprajakta, 2019-10-29:
Hello!

I will check once again the repository and upload the if anyfile is missing.

Prajakta Lokhande

DvdMgr, 2019-10-29:
It looks like the EDs in your simulation weren't requesting ACKs, so no DL messages were being sent by the GW! Adding the following line in the simulation script seems to fix the issue:

```Config::SetDefault ("ns3::EndDeviceLoraMac::MType", EnumValue(LoraMacHeader::CONFIRMED_DATA_UP));```

Prapi123, 2019-10-30:
Hello, 
Thanks for your time and efforts to look into the matter.

I added the line in complete-network-example to test  and got below error
[image]

later I tried to change Mtype by accessing EndDevice Mac layer with following code below,
`for (NodeContainer::Iterator j = endDevices.Begin ();
       j != endDevices.End (); ++j)
    {
      Ptr<Node> node = *j;
      Ptr<LoraNetDevice> loraNetDevice = node->GetDevice (0)->GetObject<LoraNetDevice> ();
      Ptr<LoraPhy> phy = loraNetDevice->GetPhy ();
      
      Ptr<LoraMac> mac = loraNetDevice->GetMac ();
      Ptr<EndDeviceLoraMac> edLoraMac = mac->GetObject<EndDeviceLoraMac> ();
      edLoraMac->SetMType (LoraMacHeader::CONFIRMED_DATA_UP);
      
    }`

for this I got this error,
[image]

running it under gdb has the following results.
[image]

I made only the changes suggested by you and nothing else.

Please help me out if you know what mistake I am making.

Regards

DvdMgr, 2019-10-30:
Strange, I don't get that error at all. I inserted the 
```
Config::SetDefault ("ns3::EndDeviceLoraMac::MType", EnumValue (LoraMacHeader::CONFIRMED_DATA_UP));
```
line right after the `CommandLine.Parse`, can you try that?

Prapi123, 2019-10-30:
Hello, 

I tried it, It still gives the same error as above.

Is it because of my changes in lora-helper.cc and lora-helper.h?

DvdMgr, 2019-10-30:
I added the line on the adr branch on your repository (last change was 1 hour ago), so I should have the same code as you. Are you sure you don't have any other changes locally?

Prapi123, 2019-10-30:
I have master-branch repository on my local machine , only four files are different in this repository they are
lora-helper.cc , lora-helper.h, lora-packet-tracker.cc, lora-packet-tracker.h. These files are from adr-branch.  which are exactly the same as in my repository.

Prapi123, 2019-10-30:
There is also one other issue, I just identified, when I used only adr-branch repository and tried to run test, it says the test is failed. 
Previously I had master repository and 4 file that I mentioned before were from adr-branch.

DvdMgr, 2019-11-05:
Regarding the test failures, this was because the wrong interference matrix was used - if you rebase on the latest changes, it should all work!

Prapi123, 2019-11-12:
Hello,

This issue is resolved and the methods created works fine for me. You can close this issue if you want.

I faced one issue while creating a method for downlink packets.
When I extracting address of endDevices.  with the code below

` for (NodeContainer::Iterator j = endDevices.Begin ();
       j != endDevices.End (); ++j)
    {
      Ptr<Node> node = *j;
      Ptr<LoraNetDevice> loraNetDevice = node->GetDevice (0)->GetObject<LoraNetDevice> ();    
      Ptr<LoraMac> mac = loraNetDevice->GetMac ();
      Ptr<EndDeviceLoraMac> edLoraMac = mac->GetObject<EndDeviceLoraMac> ();
      LoraDeviceAddress edAddress = edLoraMac->GetDeviceAddress ();      
    }`
 it worked only before  Simulator::Run (); Simulator::Destroy ();

when the above piece of code was placed after Simulator::Run (); Simulator::Destroy ();

it gave error as
[image]

That is why there was some difficulties creating methods in lora-packet-tracker for downlink packet.
If possible, can you please explain me why this is happening.

Thanks

DvdMgr, 2019-11-18:
Have you tried putting the code before `Simulator::Destroy ()`?

closed 2019-11-19 by DvdMgr