signetlabdei/lorawan #41: Setting different data rate for different set of end device

shobhuagg1992, 2019-10-11:
I am trying to configure different end devices with different values of spreading factors but it seems like SF can be configured using datarate. I am quite new with NS-3
Can you give a simple illustration of how to use the SetDataRate method for it?

Thanks

DvdMgr, 2019-10-14:
Hi, in the EU868 region, there's a direct correspondence between Data Rate and Spreading Factor: DR0 = SF12BW125, DR1 = SF11BW125, ..., DR5 = SF7BW125. 

You can find more about this correspondence on the Regional Parameters document of the LoRaWAN standard!

shobhuagg1992, 2019-10-14:
Thanks for your response. I made some modifications in the helpers and I am able to set different spreading factors for a group of end devices.

EduLima7, 2019-10-29:
In the EU868, you can do it using `LoraMacHelper::SetSpreadingFactorsUp` method. This method sets the Data Rate from the sensitivity of each End Device. For example, on `complete-network-example` line `macHelper.SetSpreadingFactorsUp (endDevices, gateways, channel);` sets Data Rate of `EndDevices` NodeContainer.

shobhuagg1992, 2019-10-29:
Thanks. I was able to configure different SF for different nodes.

closed 2019-10-29 by DvdMgr