signetlabdei/lorawan #44: Rough Number of Nodes that can be simulated

Prapi123, 2019-10-28:
Hello,

I am concerned to know the maximum number of nodes that can be simulated with this module.
I am trying to simulate around 30k of nodes. 
Is it possible to simulate these number of nodes?

DvdMgr, 2019-10-29:
This depends on several factors:
- Your hardware's RAM: despite the fact that the simulated LoRaWAN stack is lightweight, 30k nodes are a lot, and may become a burden on the system, risking out of memory errors;
- The traffic the nodes generate: if you simulate very sporadic traffic and memory is not an issue on your system, simulating 30k nodes may be possible. The highest number of nodes I've tested the simulator with, though, is 10k, and this was with an older and simpler version of the simulator;
- Confirmed traffic will have a considerable effect on simulation time, since re-transmissions act as multipliers of baseline traffic, particularly so if the network is stressed and a high number of re-transmissions is configured and then employed by the nodes. 
- The length of the simulation is also a factor you should keep in mind: simulating one day is very different than simulating just a few minutes. When setting the simulation length, you need to make sure you are not capturing any transient effects (transmissions at the very beginning of a simulation, for instance, will have better performance since re-transmission traffic has not yet had the chance to build up). Note that the PacketTracker allows you to compute metrics skipping initial and final transients. 

All in all, the best advice I can give is to simulate by steps: maybe start with a lower amount of nodes generating light traffic, take note of how the simulator performs and then increment the number of nodes and traffic based on the results you get, always checking that everything is working as expected - if you get abysmal performance with 10k nodes, for instance, it may be useless to simulate 30k devices. 

If you get any insight on scalability or have further questions, feel free to share - this is an area I'm also interested in!

Prapi123, 2019-10-29:
Thanks you for Answering. 
I have Simulated till 10k nodes with the current version. The duration of the simulation was set for 12 hours. It took its own sweet time to complete. I am using a google server for this purpose.
I am following the same strategy as you mentioned, increasing number of packets step by step. 
I will keep you posted about my results.

Regards

Prapi123, 2019-11-26:
Hello,

You can close this issue from my end.

Thanks

closed 2019-11-26 by DvdMgr