signetlabdei/lorawan #45: Capturing Downlink packets

kaustubhshrotri, 2019-10-28:
Hello All,

I want to capture  downlink packets during a simulation.
Is there a way to do so?

I am using Complete-network-example.cc for my simulation with modifications.

Thanks

DvdMgr, 2019-10-29:
This sounds similar to #40 - can you elaborate on what you need, more precisely? Maybe we can coordinate with @Fundoprajakta to make it happen.

kaustubhshrotri, 2019-11-12:
Hello,

My issue is to detect in which receive window the downlink packets are received.
is there a way?

DvdMgr, 2019-11-18:
As of now, we do not save this information. However, you can easily customize the `LoraPacketTag` class to memorize the number of the receive window that was used in the reception of the ACK, modify the packet's tag upon reception and retrieve the information later, for instance from within the `LoraPacketTracker`

closed 2019-12-13 by DvdMgr