signetlabdei/lorawan #49: Tests, tests, tests!

DvdMgr, 2019-11-05:
Tests are a vital component of the module, and a great way to get started if you want to understand what's going on under the hood. 

As of now, we have tests for the PHY layer model implementation, but more is needed for the MAC layer and for Network Server functionality.

If you want to contribute towards this, you are welcome! Please feel free to use this issue to ask questions on how to get started - I will be happy to assist you.

bolasoma, 2020-09-08:
Hello David,

I would like to know if it is possible to add a LDAP authentication to this module, any guidance on how to get started will be very welcome.

Cheers,

Soma

gaurivn, 2021-03-15:
@DvdMgr, I would like to work on this issue. Could you please advise me on the same

DvdMgr, 2021-03-16:
Sure! A good starting point would be to make sure tests run correctly, both on the master and on the develop branches. If this is not the case, we need to identify the test that does fails and understand why. You can follow the readme instructions to compile the module and run the tests!

gaurivn, 2021-03-16:
Thank you Sir for replying. I will do accordingly.

gaurivn, 2021-03-16:
> Sure! A good starting point would be to make sure tests run correctly, both on the master and on the develop branches. If this is not the case, we need to identify the test that does fails and understand why. You can follow the readme instructions to compile the module and run the tests!

The lorawan test suite passed on the develop branch with no errors, but on master branch, it gave errors for lorawan/model/building-penetration-loss.cc, please find the screenshot attached below:
[image]

DvdMgr, 2021-03-17:
Ok, as I expected, it's failing due to a change in the ns-3 APIs - the IsOutdoor method was deprecated, !IsIndoor() should be used instead. Can you fix it? I'll be happy to merge your pull request!

gaurivn, 2021-03-17:
> Ok, as I expected, it's failing due to a change in the ns-3 APIs - the IsOutdoor method was deprecated, !IsIndoor() should be used instead. Can you fix it? I'll be happy to merge your pull request!

I used !IsIndoor() instead but I am still getting more errors, I will try fix those..please find the screenshots attached below:
[image]
[image]

DvdMgr, 2021-03-18:
It looks like the problem is that in the tests we are trying to compare an instance of ns3::Time with an integer. This is happening inside the NS_TEST_EXPECT_MSG_EQ macro - it should be enough to change that 0 to a Seconds(0), so that it becomes a Time instance!

gaurivn, 2021-03-18:
Thank you, I made the change and the one error remaining was due to deprecated method - MakeMobilityModelConsistent() in complete-network-example.cc
[image]
I commented it out and committed again and the lorawan test suite passed without errors.
[image]

closed 2021-11-19 by DvdMgr