signetlabdei/lorawan #51: Changing bandwidth

philipskotte, 2019-11-18:
Hi! 

I am simulating the simple-network-example.cc, but I can't figure out how to change the bandwidth for the simulation. Right now it is using the bandwidth of 125000 Hz, but I want to change that to 250000 Hz.
Do you have any suggestions how to do this ?

DvdMgr, 2019-11-18:
Hi,
to do that, you can change the configuration of channels in LorawanMacHelper. Keep in mind these two things:
- Our error model was developed for transmissions at 125kHz, and does not guarantee correctness in case a 250kHz bandwidth is used.
- Our model does not take into account overlapping channels: you should space your 250kHz channels appropriately, as to have them not overlap.

Time on air computation with a 250kHz channel should be taken care of appropriately by the respective functions, no need to tweak that.

philipskotte, 2019-11-19:
Hi again! 
Thank you for the answer! 
I just have one more question and that is in which file do you implement the size of the package that will be sent?

DvdMgr, 2019-11-19:
This is decided by the application layer - typically, in the examples we
use the PeriodicSender class.

DvdMgr, 2019-11-30:
I'm closing this, feel free to keep commenting if you need further assistance!

closed 2019-11-30 by DvdMgr