signetlabdei/lorawan #52: How to measure Throughput of network?

shobhuagg1992, 2019-11-21:
Hello
I am trying to measure the throughput of the network. Is there a way to do that?

Thanks

DvdMgr, 2019-11-24:
Hi! Do you mean throughput in terms of bits/s or in relation with offered traffic, so as to make comparisons with an Aloha access scheme?

shobhuagg1992, 2019-11-24:
Yes in terms of bits/second.

DvdMgr, 2019-11-27:
As of now, not out of the box. The most straightforward way to do this would be to hook a function to the `ReceivedPacket` trace source of the Gateways - this way you could log into a file the time at which you receive the packets and their length in bytes, and compute the throughput either online directly during the simulation or offline in a second phase!

closed 2019-12-13 by DvdMgr