signetlabdei/lorawan #54: Extract Class A End Devices into their own function

peggyopal, 2019-11-22:
The title says it all! 

This way it would make it really easy to create new classes for Class B and C devices to extend from a base `EndDeviceLorawanMac` class.

DvdMgr, 2019-12-04:
I merged the PR relative to this issue, I feel like we can close this unless you'd like to have further discussion!

peggyopal, 2019-12-04:
I have to agree with you! 

Also, I'm sorry about the error that happened. I thought I combed through the logs enough, apparently I did not! Would you mind sharing how you found it? :)

closed 2019-12-04 by peggyopal

DvdMgr, 2019-12-05:
I ran `network-server-example` and saw that the second device, which should require an ACK, tried to send the message 8 times. The Network Server correctly received the message, and the Gateway was sending replies, however I saw from the log that the `Close(First/Second)ReceiveWindow` calls happened right after `Open(First/Second)ReceiveWindow`, and thus did not allow the PHY to lock on the incoming packet. I added logging to show the value of `m_receiveWindowDurationInSymbols`, which is used to compute the delay at which to close the window, and saw that it was zero, thus explaining the non-existent delay between opening and closing of the receive windows. It was enough to initialize the variable to get it to work correctly!