signetlabdei/lorawan #58: Create compiled documentation

DvdMgr, 2019-11-30:
As of now, we have a `lorawan.rst` file that describes some of the functionalities of the module, but this is not enough. I've identified the following tasks to improve the documentation of the module: 
- Add a compiled documentation page, possibly hosted through Github Pages for immediate access by users.
- Add documentation on how the Network Server works
- Add description of tests, underlying models and validation
- Add a page describing how the simulator was validated by comparison with a mathematical model

KishkinJ10, 2021-03-21:
hello everyone , i am a beginner and i want to work on this issue can someone tell me how should i start .

DvdMgr, 2021-03-22:
Hi! A pre-requisite to start working on this issue is a somewhat deep knowledge of both the LoRa modulation and the LoRaWAN protocol, in addition to an understanding of how these components are modeled in ns-3. If you have these attributes, feel free to start tackling one of the listed bullet points.

closed 2021-11-19 by DvdMgr

wilsongus2003, 2021-11-28:
Good afternoon, where can I get information on how to generate pcap files, oh how to track package traces?