signetlabdei/lorawan #60: I need to analyze the security features of LPWAN technologies.

OniremE, 2019-12-08:
Is it possible use this module to analyze the security parameters of LPWAN technologies?

DvdMgr, 2019-12-09:
Currently, we don't model any of the security features - what exactly are
you looking to simulate?

OniremE, 2019-12-09:
I want to analyze the encryption feature and find the possible vulnerabilities in LoraWan and Sigfox.

DvdMgr, 2019-12-09:
In this case, I don't think this is the right tool: since we focus on analyzing the scalability of the network, we completely avoid modeling encryption to save computation time!

OniremE, 2019-12-09:
could you give me a suggestion for me?

DvdMgr, 2019-12-09:
I'm sorry, I don't have any experience with the security side of things in LoRaWAN - Google Scholar might be a good starting point!

closed 2019-12-09 by DvdMgr