signetlabdei/lorawan #61: Packet Size of Reply Packet

Prapi123, 2019-12-09:
Hello,

When I set MType :: Confirmed Data, 
I want to send packets to end devices with a particular packet size. 

It would be helpful for me if you suggest how can I achieve this.

Thanks and Regards

DvdMgr, 2019-12-11:
If you want a fixed packet size, you can directly change the code in `EndDeviceStatus::GetCompleteReplyPacket` to create a packet with a set payload size!

Prapi123, 2019-12-15:
Hello @DvdMgr 

I have been modifying EndDeviceStatus.cc file for my convenience and I am facing some  issue in it.
1.  
I have used a code to modify  replypaylaod size as given below
` for (NodeContainer::Iterator j = endDevices.Begin ();
       j != endDevices.End (); ++j)
  {
    Ptr<Node> node = *j;
    uint32_t senderID = node->GetId ();
    std::cout << "senderID: " << senderID << std::endl;
    Ptr<LoraNetDevice> loraNetDevice = node->GetDevice(0)->GetObject<LoraNetDevice> ();
    Ptr<LoraPhy> phy = loraNetDevice->GetPhy ();

    Ptr<LoraMac> mac = loraNetDevice->GetMac ();
    Ptr<EndDeviceLoraMac> edLoraMac = mac->GetObject<EndDeviceLoraMac> ();
    LoraDeviceAddress Address = edLoraMac->GetDeviceAddress();
    Ptr<EndDeviceStatus> edStatusP3 = CreateObject<EndDeviceStatus> (Address, edLoraMac);

    Ptr <Packet> replypayload = Create<Packet> (126);
    edStatusP3->SetReplyPayload (replypayload);

    LoraFrameHeader d = edStatusP3->GetReplyFrameHeader();
    std::cout<< d << std::endl;
  }`

Above I am changing the attaching the payload with Setreplypayload() but still I can see output as below 
[image]

It is creating an empty packet instead of creating a packet with size I have already stated.

My idea is to emulate handshake between  NS and end device and so I want to create reply packet according to the size of  uplink packet.
for example if uplink packet size is 30 byte then downlink packet should be 23
                    if uplink packet size is 50 byte then downlink packet should be 15

Thanks and Regards
@Prapi123

DvdMgr, 2019-12-16:
In that case, you have two ways to go about it:
1. Undo your changes to `end-device-status.cc`, and create a new `NetworkControllerComponent` object - this will need to implement a `BeforeSendingReply` method, which is a callback called by the `NetworkController` before sending a reply, and is where you will have the chance to modify the `EndDeviceStatus`'s reply payload by calling the `SetReplyPayload` method. You can use the components defined in `network-controller-components.cc` file to draw some inspiration about the structure and functionality. Don't forget to add it to the `InstallComponents` method of the `NetworkServerHelper` to activate it!
2. Define all the needed logic directly inside the `GetCompleteReplyPacket` method: you can remove the if/else that either uses the m_reply.payload or creates a new packet from scratch, and make it so that it always creates a packet with the needed size. If you want to always send a packet with a payload of 50, for instance, you simply use `replyPacket = Create<Packet> (50);`. 

The code you mentioned above does not work because you are creating new `EndDeviceStatus` objects, instead of acting upon the ones that already exist and are used in the simulation!

Prapi123, 2019-12-19:
Hello,

Thanks for responding to this issue.
I could emulate TLS handshake with above recommended process. 

I am trying to solve minor error, I want to create a packet  replyPacket = Create<Packet> (1200);
if gives an error "large integer implicitly truncated to unsigned type [-Woverflow] "

I suppose this is due to range of Unsigned integer  which is 0-255 so tried to change data-type in source file. Still no luck.

DvdMgr, 2019-12-19:
> Create (1200);

Keep in mind that the maximum size for the payload of a LoRaWAN packet is
from 50 to 200 bytes, depending on the spreading factor! Sending packets
1200 bytes long just is not possible with this technology

closed 2020-01-09 by DvdMgr