signetlabdei/lorawan #62: Features of Ns 3 lorawan over others

adnan21471, 2019-12-25:
Hello..I am new to Ns 3 and inet.
I am currently deciding which to choose over other meaning how is Ns3 better than flora (by features) and vice versa..Also which is easy to learn quickly.

Would really appreciate if someone could guide me..I am not focusing on machine learning, but just resource allocation problems..Have a nice day

Prapi123, 2019-12-26:
Hello, 

I have done an initial study on tools like, NS3 lorawan, flora, simpy to evaluate features and compare them. If you provide your email I can send you it via email.

adnan21471, 2019-12-26:
Hello..As I am new to GitHub, here is my email according to github

59202730+adnan21471@users.noreply.github.com

adnan21471, 2019-12-26:
Hello.
Here is my email address 
adnan21471@gmail.com

DvdMgr, 2020-01-09:
@Prapi123, if you feel like you can share your comparison publicly, I think it would be good! Contact me if you are interested. In the meantime, I'm closing this assuming the issue was resolved.

closed 2020-01-09 by DvdMgr