signetlabdei/lorawan #65: Gateway rules while transmitting downlink packet

Fundoprajakta, 2020-01-11:
Hello Again,

This not an issue but a query.
I am capturing downlink packets and calculating its packet error rate at both end device and  gateway level.
I see the downlink packet error rate is much higher as compared to uplink which is an expected behaviour.  I am curious to know how gateway is deciding which packets to drop and which packets to pass. what are the rules for downlink packet transmission.

I am using adr branch. I checked network-controller-components.cc/h and network-scheduler.cc already to search for the rules, still not close to what I was searching.

Is there a particular class or classes I should look for?

Thanks and Regards

DvdMgr, 2020-01-13:
How are you computing the downlink packet error rate? If you compute it as the
number of correctly received DL packets divided by the number of confirmed UL
packets, yes, it is expected that it should be low, when the number of devices
requesting an ACK is high or there are few GWs.

If, on the other hand, you are computing the downlink packet error rate as the
number of correctly received DL packets divided by the number of sent DL
packets, I would not expect this to be lower than the UL packet success rate.
Can you confirm you are using the former method of computation?

As for how the GW decides which packets to reply to, this is done in the
`NetworkScheduler::OnReceiveWindowOpportunity` function: when a receive window
is opened, the NS will call the `NetworkStatus::GetBestGatewayForDevice`,
which in turn will go over the list of GWs that can reach the ED the receive
window is for, and ask each one whether it is available for transmission using
the `GatewayStatus::IsAvailableForTransmission` method. In turn, this method
will check whether the GW is already booked for a transmission, whether it is
currently transmitting and whether it would be able to transmit now according to
its duty cycle.

So, in short, you could say the NS will always try to serve EDs according to a
first come first serve policy.

If you have any further questions don't hesitate to ask!

Fundoprajakta, 2020-01-13:
Hello,
Thanks for responding to the query.
I have created a downlink packet tracker m_packetTrackerDownlink in lora-packet-tracker.h/cc to record only downlink packets. 
All methods to calculate downlink PER are using this tracker. 
from  the two cases mentioned above, I am using the 2nd one, (DL received messages/DL sent messages). The packet error is very low in this case

DvdMgr, 2020-01-13:
Ok, great! Can I close this?

Fundoprajakta, 2020-01-13:
your explanation answers my first asked question.
I have another query, 
as I am using the 2nd method to calculate DL PER (DL received messages/DL sent messages). and I am getting much lower PER than UL PER. Is this because of higher number of downlink packets?

DvdMgr, 2020-01-13:
Sorry, I see I mixed up the language. 

# of DL received messages / # of DL sent messages is a DL packet _success_ rate. I expect this to be quite high for all sizes of networks. Is this what you are observing?

# of DL received messages / # of confirmed UL sent messages, instead, is what we could call a successful confirmation rate. I expect this to be high (i.e., close to 1) for very small networks, and that it should go much, much lower as the network size increases and the GW can not keep up with the high amount of ACK requests.

Is this coherent with what you are observing?

Fundoprajakta, 2020-01-13:
Yes, This is what exactly happening with network as well. Thanks for your time and efforts to explain me my query. Sorry for the confusion. 
Please close the issue.

DvdMgr, 2020-01-13:
Great to see everything is working as expected - keep asking if you have any other questions!

closed 2020-01-13 by DvdMgr