signetlabdei/lorawan #66: Send a packet to the ED at any time.

QiuYukang, 2020-01-15:
Hello, I want to ask a question, is there any way to make the LoRa NS send packet to the ED at any time? Can you give me some hints to modify the code? Thank you very much.

DvdMgr, 2020-01-15:
As of now, we only support Class A LoRaWAN End Devices - you would need a
MAC layer class implementing Class C behavior to obtain what you are
describing. Furthermore, some modifications would be needed at the Network
Server, since as of now it will only send packets in correspondence of
Receive Windows. I know there is some effort going towards this from other
people: if you are interested, we can probably coordinate with them. Let me
know!

QiuYukang, 2020-01-16:
> As of now, we only support Class A LoRaWAN End Devices - you would need a MAC layer class implementing Class C behavior to obtain what you are describing. Furthermore, some modifications would be [...]

thank you! I will have a try.

DvdMgr, 2020-01-16:
You can try contacting @peggyopal, I know she is working on something similar.

Closing this for now!

closed 2020-01-16 by DvdMgr