signetlabdei/lorawan #68: Fcnt for uplink should not be incremented for retransmission

theelbowclaps, 2020-01-22:
## Expected Behavior
Same as title.
According to lorawan spec 1.0.3 (https://lora-alliance.org/sites/default/files/2018-07/lorawan1.0.3.pdf), section **4.3.1.5 ** (ln 520-530) seems to suggest retransmitted frame's Fcnt should not be incremented. Similar idea is expressed in sect **4.3.1.1** ln 443-444.

## Actual Behavior
end-device-lorawan-mac.cc line 291:  m_currentFCnt++;

## Steps to Reproduce the Problem
N/A

## Specifications
N/A

DvdMgr, 2020-01-22:
Thanks for pointing this out. 

Actually, frame counters have not been implemented/tested so far, as their behavior does not impact the outcome of simulations we are interested in. If you want to implement them correctly also at the `NetworkServer`, please submit a pull request and I'll look into it! Otherwise, I can just remove the line you pointed out to fix this.

gaurivn, 2021-04-09:
On clicking the link provided in the issue, it says page not found, could you please provide the working link

> ## Expected Behavior
> Same as title.
> According to lorawan spec 1.0.3 (https://lora-alliance.org/sites/default/files/2018-07/lorawan1.0.3.pdf), section **4.3.1.5 ** (ln 520-530) seems to suggest [...]

DvdMgr, 2021-04-11:
Hi, searching in the LoRa Alliance website for the 1.0.3 specification as the url suggests brings up this [updated link](https://lora-alliance.org/resource_hub/lorawan-specification-v1-0-3/).

closed 2021-04-30 by DvdMgr