signetlabdei/lorawan #69: Conditional Sending of packets

Fundoprajakta, 2020-01-27:
Hello 
I am wondering how can I modify behavior of  nodes to send packet. 
My goal is to send next transmission only if the previous transmission is received by gateways.

I tried using m_macPacketTracker first, but somehow it is not working for me. 
Please suggest me this is possible and what code should I modify in order to achieve this.

Thanks and Regards

DvdMgr, 2020-01-27:
What you are describing sounds like something that can be obtained by modifying the application layer - the `PeriodicSender` class, if you are referring to the examples. 

The packet tracker is only used to keep track of the packets from a network standpoint, and it doesn't influence the behavior of single devices.

Fundoprajakta, 2020-01-28:
Is there a way to check which applications are installed on a node?

DvdMgr, 2020-01-28:
Sure - for these questions concerning ns-3, however, please refer to [the ns-3 manual](https://www.nsnam.org/docs/manual/html/index.html): let's keep all discussions in this repo focused on the lorawan module.

Fundoprajakta, 2020-02-01:
Hello I have added a piece of code in complete-network-example.cc as below

`for (NodeContainer::Iterator j = alldevices.Begin ();
       j != alldevices.End (); ++j)
  {
    Ptr<Node> nodeP3 = *j;
    for (uint32_t j = 0; j < nodeP3->GetNApplications (); j++)
    {
      Ptr<PeriodicSender> PeriodicSend = DynamicCast<PeriodicSender> (nodeP3->GetApplication (j));
      if (PeriodicSend)
      {
        PeriodicSend->StopApplication();
      }

    }
  }`

And I keep getting error as below. 

msg="You have attempted to read beyond the bounds of the available buffer space. This usually indicates that a Header::Deserialize or Trailer::Deserialize method is trying to read data which was not written by a Header::Serialize or Trailer::Serialize method. In short: check the code of your Serialize and Deserialize methods.", file=./ns3/buffer.h, line=1002

DvdMgr, 2020-02-03:
This looks like an ns-3 issue, not related to the lorawan module. Closing.

closed 2020-02-03 by DvdMgr