signetlabdei/lorawan #70: Downlink Transmission and Receive Power

Fundoprajakta, 2020-02-05:
Hello,

How can I approach to find out Downlink Transmission power and Received  Power for signal?

Thanks and Regards

DvdMgr, 2020-02-07:
Hi, are you looking for this information inside the `PacketTracker`?

Fundoprajakta, 2020-02-08:
Hello,

My Idea is to get SNR and RSSI value for downlink as well.

shobhuagg1992, 2020-02-09:
Hello Prajakta
Can you please let me know how to unable downlink traffic?

Thanks

Fundoprajakta, 2020-02-09:
Hello @shobhuagg1992 

The Downlink traffic is already enabled in the module. You have to set LoraMacheader value to confirmed_data _up. 
With Packettracker you can compute the metrics.  You can check issue no. 40 which is closed now. I have discussed with @DvdMgr  about this earlier. You can also visit my profile and check lorawan repository which I forked from here to draw some inspiration from the changes that I have made.

Thanks

shobhuagg1992, 2020-02-10:
Awesome. Thank you so much, will try that.

closed 2020-04-04 by DvdMgr