signetlabdei/lorawan #71: EndDeviceStatus::GetSecondReceiveWindowFrequency() error

Fundoprajakta, 2020-02-06:
Hello,
i have used below small code to extract Second receive window frequency In network server example.

`for (NodeContainer::Iterator j = endDevices.Begin ();
       j != endDevices.End (); ++j)
  {
    Ptr<Node> node = *j;

    //uint32_t senderID = node->GetId ();
    //std::cout << "senderID: " << senderID << std::endl;
    Ptr<LoraNetDevice> loraNetDevice = node->GetDevice(0)->GetObject<LoraNetDevice> ();
    Ptr<LoraPhy> phy = loraNetDevice->GetPhy ();

    Ptr<LoraMac> mac = loraNetDevice->GetMac ();
    Ptr<EndDeviceLoraMac> edLoraMac = mac->GetObject<EndDeviceLoraMac> ();
    LoraDeviceAddress Address = edLoraMac->GetDeviceAddress();
    NetworkStatus NetStatus;
    Ptr<EndDeviceStatus> edstatus = NetStatus.GetEndDeviceStatus(Address);
    double windowfrequency = edstatus->GetSecondReceiveWindowFrequency();
    std::cout << windowfrequency << std::endl;
    
  }`

I am getting below error. I am confused why is this happening
[image]

Any method used from EndDeviceStatus class gives the same error.

Can you please guide me on this.

Thanks and Regards

DvdMgr, 2020-02-06:
Since you are creating a new `NetworkStatus` from scratch, it will be empty. You should use the `NetworkStatus` belonging to a `NetworkServer` configured through the helper to get any meaningful contents in the `EndDeviceStatus` objects.

Fundoprajakta, 2020-02-06:
Hello,
Thanks for the reply.
It worked. You can close this issue

closed 2020-02-06 by DvdMgr