signetlabdei/lorawan #72: A question about the ReceiveWindow

QiuYukang, 2020-02-07:
Hello,
I run the `example/network-server-example.cc`, through logs i see this:

[image]
This means the ED received an ACK after closing the first receive window(the second receive window has not been opened at this time).
Finally the second receive window was not opened either.
**why this is happening?** 
From the source code `lorawan/model/class-a-end-device-lorawan-mac.cc` it seems that the second receive window will still should be opened, I didn't find where you called `m_secondReceiveWindow .Cancel()`:
```c++
void
ClassAEndDeviceLorawanMac::TxFinished (Ptr<const Packet> packet)
{
  NS_LOG_FUNCTION_NOARGS ();

  // Schedule the opening of the first receive window
  Simulator::Schedule (m_receiveDelay1,
                       &ClassAEndDeviceLorawanMac::OpenFirstReceiveWindow, this);

  // Schedule the opening of the second receive window
  m_secondReceiveWindow = Simulator::Schedule (m_receiveDelay2,
                                               &ClassAEndDeviceLorawanMac::OpenSecondReceiveWindow,
                                               this);
  // // Schedule the opening of the first receive window
  // Simulator::Schedule (m_receiveDelay1,
  //                      &ClassAEndDeviceLorawanMac::OpenFirstReceiveWindow, this);
  //
  // // Schedule the opening of the second receive window
  // m_secondReceiveWindow = Simulator::Schedule (m_receiveDelay2,
  //                                              &ClassAEndDeviceLorawanMac::OpenSecondReceiveWindow,
  //                                              this);

  // Switch the PHY to sleep
  m_phy->GetObject<EndDeviceLoraPhy> ()->SwitchToSleep ();
}
```
Can you please guide me on this.

Thanks and Regards

DvdMgr, 2020-02-07:
Hi! If the first window is successful, the second receive window is not opened (this behavior is defined in the standard). The call to cancel is in the `ClassAEndDeviceLorawanMac::Receive` method, and it will only be invoked if the received message is intended for the device. You can find it at line 176 of the `class-a-end-device-lorawan-mac.cc`!

QiuYukang, 2020-02-08:
I found it with your help, Thank you very much!

closed 2020-02-08 by QiuYukang