signetlabdei/lorawan #73: Few Questions: Downlink, Transmit power, US conf. and Gateway

shobhuagg1992, 2020-02-09:
I have a few questions regarding the simulator:

1. The documentation says the gateway can receive simultaneously on all channels. Does that mean if we configure the gateway with 8 channels, then will it be able to receive 8 (channels) X 6 (SFs)= 48 packets simultaneously?

2. I saw some people asking questions about the downlink communication. Is it available in the module? or in other terms how can we turn on the acknowledgments or downlink traffic?

3. I am trying to put a test node in the network is there a way by which I can just print the performance matrice for that test node?

4. Is the module upgraded to use US configurations?

5. Also, is it possible to change the transmit power of the end devices?

I am a student trying my hands on this technology. I would also love to contribute to this module.

Thanks.

DvdMgr, 2020-02-11:
1. The gateway has 8 rf chains, that can listen to any incoming packet, and must be centered each on a certain frequency. This means that the GW is able to listen to up to 8 packets incoming simultaneously - by default, since in an EU deployment we have 3 default channels, we place 3 rf chains at 868.1 MHz, 3 at 868.3 and 2 at 868.5, employing them all. If you have 8 distinct frequencies, the 8 rf chains can be configured to listen to each channel. If you are curious, this behavior is implemented in the `GatewayLoraPhy` class, and we represent the rf chains through objects called `ReceptionPath`. The `LoraPhyHelper` class is typically used to configure this behavior.
2. Yes, DL is available in the module. To have acknowledged traffic, you should tell your `EndDeviceLorawanMac` objects to employ confirmed traffic, and you can set this up through the `MType` attribute. Remember that in order to have the gateways deliver acknowledgments you should also set the `NetworkServer` up, this is shown in the examples anyway.
3. So far we have been mainly interested in network-wide performance. If you want to see the performance of a single node you can either set up your own callbacks listening for rx drop, transmission, etc. events or you can write your own custom function inside `PacketTracker`. This is an object that keeps track of all packets in the network, so your device's packets will be there too.
4. Some support for the US configuration was added recently, however we are keeping it separate in the us915 branch since we haven't had much time to test it yet. If you are interested in trying it out and doing a bit of test driving, we can definitely include it in the main branch. 
5. If you are interested in doing this "the realistic way", the transmission power on end devices can be changed through ADR. As of now, it seems like the `EndDeviceLorawanMac` class does not include trnasmission power as an attribute. If you are interested in contributing to the module, this could be a relatively easy first step to take!

shobhuagg1992, 2020-02-11:
Thanks for responding.
I am still confused regarding the simultaneous reception. Lets say we have a single reception path say 868.1MHz and there are 3 packets arriving all on the same channel 868.1 MHz but with different spreading factors say 7, 10 and 12. Will the gateway be able to receive all 3 packets (based on quasi-orthogonality of spreading factors) or will it just receive one of the packet and drop others? if dropped will those dropped packets be considered as "Interfered" or "No receivers"?

DvdMgr, 2020-02-11:
Each reception path can only lock on a single packet, so in your example it would lock onto the first one and drop the others. The dropped packets will be considered as lost because no more receivers are available. If you wanted them all to be received, you would need three reception paths listening at 868.1 MHz!

shobhuagg1992, 2020-02-11:
So when will the packets be considered as Interfered?

DvdMgr, 2020-02-11:
If the GW can lock onto them, they will be either correctly received, interfered, or lost due to GW transmission if the reception needs to be interrupted by a DL communication. If they arrive under sensitivity, they are lost even if the GW has some available reception paths, and marked as lost because under sensitivity. If they arrive with a power that is above the required sensitivity, and the GW cannot lock onto them, they will be considered lost because no more reception paths are available.

Note that overlapping packets using the same SF are not necessarily lost - if one is 6dB above the others, the interference matrix allows the more powerful one to be correctly received. If this does not happen, however, yes, they will be considered as interfered, provided both were locked-on by a reception path.

shobhuagg1992, 2020-02-11:
Perfect. That answers my question. 
Thanks for your help and I will definitely try to add the transmit power as attribute. I have forked the repository and will let you know once I make some contribution.

DvdMgr, 2020-02-11:
Great, thank you! I'm closing this for now then.

closed 2020-02-11 by DvdMgr