signetlabdei/lorawan #74: Out of GetBestGatewayForDevice() does not match to address of Gateway nodecontainer

Fundoprajakta, 2020-02-09:
Hello,
I have extracted device gateway address using this below code
`for (NodeContainer::Iterator j = gateways.Begin ();
       j != gateways.End (); ++j)
  {
    Ptr<Node> node = *j;
    //uint32_t senderID = node->GetId ();
    //std::cout << "senderID: " << senderID << std::endl;
    Ptr<NetDevice> NetDevice = node->GetDevice(0);
    Address addr = NetDevice->GetAddress();
    std::cout << addr << std::endl;
  }`

And used NetworkStatus::GetBestGatewayForDevice()  method to get Gateway address. 

Both of them produce different output: 

[image]

Please suggest where I am going wrong. The basic idea is to get the Ptr<Node> corresponding to NetworkStatus::GetBestGatewayForDevice()

DvdMgr, 2020-02-11:
The address returned by `GetBestGatewayForDevice` is the address of the `PointToPointNetDevice` connecting the GW to the NS, so it is expected you will get different results. If you want to get a `Ptr` to the best GW, you should be able to get them from within the `NetworkStatus` object: this object holds a map linking the addresses returned by `GetBestGatewayForDevice` to a `GatewayStatus` object, which in turn contains a pointer to the GW's Mac. You can see this is used in `src/lorawan/model/network-status.cc`, line 157!

DvdMgr, 2020-02-12:
Fixed in #76

closed 2020-02-12 by DvdMgr