signetlabdei/lorawan #80: Increase in packet loss due to interference when multiple paths are listening to same channel

shobhuagg1992, 2020-02-25:
I tried simulations using the following configuration with each device sending 100 packets:

Case1: Only one Rx path enabled for 868.3 MHz; Fixed SF=7; Number of devices=1000; Range=1700m
-packets interfered: 4800

Case2: Two Rx path enabled for 868.3 MHz; Fixed SF=7; Number of devices=1000; Range=1700m
-packets interfered: 9100

Case3: Three Rx path enabled for 868.3 MHz; Fixed SF=7; Number of devices=1000; Range=1700m
-packets interfered: 9500

I am experiencing the number of packets lost due to interference getting increased as I am increasing the number of Rx. paths for the same channel but according to the documentation if one path gets locked to one of the packets and if another packet arrives for the same channel. The only the available paths should try to lock on it so it shouldn't be interfering with the first packet, but why I am experiencing an increase in interfered packets? I am misunderstanding something here?

If we talk in terms of SX1301. On the receiving end is it kind of a bus system connected to parallel demodulators where all the parallel lines get the same data. Hence if there is an overlapping of packets occurs at the receiving antenna, all the active demodulators will be getting the same overlapped packets? Hence, all demodulators will show interference? If that is so then number of interfered packets should just double when using 2rx paths on the same channel as compared to one rx path, but the simulation results doesn't reflect that. 
Kindly clarify.

DvdMgr, 2020-02-25:
Can you share the script you used to get these results, so that I can replicate and look into them?

DvdMgr, 2020-02-25:
The first explanation that comes to mind, however, is that interference is not the only way to lose packets. In order for the simulator to catalogue a packet as lost due to interference, there first needs to be a receive path that is available to lock on the packet. So, I expect that if you sum packets lost due to interference and due to not having an available receive path, you should get the same value for all three simulations.

shobhuagg1992, 2020-02-25:
" In order for the simulator to catalog a packet as lost due to interference, there first needs to be a receive path that is available to lock on the packet."

But if the receive path is not available the packet will be lost due to no receive path available.

shobhuagg1992, 2020-02-25:
Let me commit the changes to the script and then will let you know.

DvdMgr, 2020-02-25:
Sure! It would help if you could post, fo each value of the number of receive paths:
- The number of successful packets
- The number of packets lost due to interference
- The number of packets lost due to there being no more receivers

shobhuagg1992, 2020-02-25:
Please check the attached sheet for the complete simulation results
[Simulation Data.xlsx](https://github.com/signetlabdei/lorawan/files/4252088/Simulation.Data.xlsx)

DvdMgr, 2020-02-25:
Ok, I'd say that these results look fine to me. Keep in mind that, in order to have a packet that is lost to interference, you first need to have a rx path that can lock onto it. Unless this condition is verified, we lose the packet due to not there being enough rx paths - but it's still lost! It's just a matter of how we count things.

If we only look at the case with 1000 devices:
- For 1 rx path, you have 7600 packets that are lost because no receive paths are available. This is to be expected, since some packets will find the GW already occupied with the reception of another packet. Even when packets are correctly locked-on by the rx path, they might be lost because of interference.
- For 2 rx paths, you get a decrease in the number of packets that find a fully occupied gw and are lost because of no more rx paths, as expected. Now, since more packets can be accepted by the GW and then be lost due to interference, you get an increase in interference.
- Finally, when you have 3 rx paths the GW can initiate reception of all packets, and you get even more packets that can potentially be lost because of interference.

shobhuagg1992, 2020-02-25:
Okay so correct me if I am wrong.
In case of one receive path, out of 7600 packets which were dropped due to no receive paths available most of those packets were bound to be interfered. The gateway is not able to process them hence they don't show up as interfered. But when we open 2 receive paths 7100 packets out of 7500 packets were able to be locked on to the receiver and hence after processing the gateway declare them as interfered?

DvdMgr, 2020-02-25:
Yes, that is correct.

shobhuagg1992, 2020-02-25:
But still the proportion of packets dropped due to interference with 2 receive paths is way more as compared to when one receive path is active.

DvdMgr, 2020-02-25:
Proportion of interfered packets with respect to what, the total amount of packets?

shobhuagg1992, 2020-02-25:
You see when one rx path is active 
We sent total 100000 packets out of which 92400 packets were able to lock on to the receiver and out of those 4800 packets were interfered. Which means 4800/92400= 0.0519
but when we open 2 paths then out 100000 packets 99500 packets were able to lock on the receivers and out those 9100 packets were interfered. which means 9100/99500= 0.0915

As the whole network is transmitting data at fixed period, the proportion packets interfering should be the same, right?

DvdMgr, 2020-02-25:
I see what you mean. Are these simulation runs randomized and/or averaged?

shobhuagg1992, 2020-02-25:
What is meant by that?

DvdMgr, 2020-02-25:
How many simulations did you perform to obtain these numbers? Or are they single-shot simulations? Were they performed with the same `--RngRun` parameter?

shobhuagg1992, 2020-02-25:
All these 3 cases performed separately.

DvdMgr, 2020-02-25:
Can you repeat both simulations for 1 and 2 rx paths, but give them both `--RngRun=10` as a parameter?

shobhuagg1992, 2020-02-25:
What does that do? And can how to use it?
Sorry I am new to this simulator

DvdMgr, 2020-02-25:
If you usually run simulations using `./waf --run simulation-name`, try running them with `./waf --run "simulation-name --RngRun=10"`. This will (more or less) change the random seed the simulator is using, and should give slightly different results. Once we get these results, we can look at the variation with respect to your original results and this might help us decide whether this effect is actually happening or can be catalogued as variability between different simulations.

shobhuagg1992, 2020-02-25:
Awesome. I get it.
Will get back to you once I have those results.

shobhuagg1992, 2020-02-25:
Got the same results. 
1 rx paths= 100000 87600 4800 7600 0 0 
2 rx paths= 100000 90400 9100 500 0 0

DvdMgr, 2020-02-25:
Are you rounding the results? If you get the exact same values even if you change the `RngRun` parameter something's off.

shobhuagg1992, 2020-02-25:
I didn't change anything in any file except the lorawan-mac-helper.cc
That too just for changing the number of receive paths.

DvdMgr, 2020-02-25:
Ok, I think I'd like to take a look myself then. Can you commit the changes to your repo? Maybe provide the code for both cases of 1 and 2 rx paths, and comment one of them so I can replicate this more easily.

shobhuagg1992, 2020-02-26:
I have committed the changes in my repo. 
The code for running is in examples named "complete-network-example.cc"
The only other changes were made in "lorawan-mac-helper.cc"

I have commented on things to make it clearer. Please let me know if you need anything from my side.
Hope to listen from you soon.

DvdMgr, 2020-02-27:
I've looked into this a little bit. To me, it seems results make sense, since the vast majority of packets that are lost because of no more receivers would be lost because of interference anyway: in other words, the no more receivers figure is hiding interfered packets, as we discussed earlier. I've come up with a simple example that I think illustrates quite well why it looks like interference is growing.

Assume we are in a 1 rx path scenario, and we send 4 packets in total: of these, 2 are successful, one is marked as lost because no more receivers but would have been lost anyway because of interference, and the last packet is lost, too, because of the interference from the packet we weren't able to lock onto. In this case, the computation you gave above gives the following results, where I is the number of packets marked as interfered, T is the total number of sent packets, and N is the number of packets marked as lost because No More Receivers:
 I / (T - N) = 1 / (4 - 1) = 0.3333

Now, let's look at the scenario with 2 rx paths: this time, we have the same 2 successful packets as before, and since we have 2 rx paths we can lock onto both the other packets that collide. This time, too, they collide, and are thus marked as interfered. The computation now gives:
I / (T - N) = 2 / (4 - 0) = 0.5

As you can see, this is the effect we are seeing in simulation too, with an apparent increase in interference. However, I believe this is an unfortunate consequence of the way we mark packets, that can be confusing and is not very intuitive.

DvdMgr, 2020-02-27:
Also, another tip: you can get significantly shorter simulations that yield exactly the same results if you reduce the simulation time by a factor of 100: the applications that are installed on the EDs are of class `PeriodicSender`, meaning they will transmit with a fixed period. This will give you the same collisions and successful packets over and over, for each application period - this also explains why you always got multiples of 100 as your results!

shobhuagg1992, 2020-02-28:
Thanks for your time and explanation. Its the marking of the packets that caused this confusion.

DvdMgr, 2020-02-28:
I agree it can be confusing - unfortunately, computing whether a packet that found no available receiver would be lost due to interference would slow simulations down. If you are interested in tackling this, we could try and implement an optional "interfered regardless of rx paths" metric. Let me know! 

Closing this in the meanwhile - and again thank you for bringing this up, it might help others in the future!

closed 2020-02-28 by DvdMgr