signetlabdei/lorawan #81: CAPTURING PACKETS

OniremE, 2020-03-05:
Hi. is it possible to sniff packages during simulation?

DvdMgr, 2020-03-06:
If you need a list of all packets that were sent during the simulation, you can access that through the `LoraPacketTracker` object!

OniremE, 2020-03-08:
I need to analyze the packets traffic, something similar to a packet sniffer with wireshark, but I have read that this module does not implement pcap.  Is there any way to do so?

DvdMgr, 2020-03-16:
As of now, we don't support pcap as you said. You can use trace sources to access individual packets, and log the contents of their headers and payload to files for instance. If you are interested in contributing pcap logging helpers, I can definitely guide and help, let me know!

DvdMgr, 2020-04-04:
Closing due to inactivity.

closed 2020-04-04 by DvdMgr