signetlabdei/lorawan #91: AES encyrption and decryption using NwkSKey and AppSKey

arlidaabdullah, 2020-08-13:
Hi 

I'm trying to reproduce the LoRaWAN state-of-the-art communication where AES_CMAC and AES_CTR session keys are inserted to do the encryption and decryption, and eventually propose a new enhanced crypto algorithm. Can you guide me on where to start and which phase should the algo be inserted?

Thanks and regards,

DvdMgr, 2020-08-25:
Hi!
As you probably know, we currently don't perform any encryption - usually, system-level network simulators make some assumptions to simplify the underlying layers, and crypto is one of the first things to go. 

I think we need to better understand what you mean with crypto algorithm - is this related to key exchange, or the authentication procedure, or does it involve message exchange in any way? In this case, I think it makes sense to implement the procedures and show any improvements!

DvdMgr, 2021-03-22:
Closing due to inactivity.

closed 2021-03-22 by DvdMgr