signetlabdei/lorawan #94: Flow Monitor integration

Jean-mrs, 2020-10-19:
Hello, i need to carry out studies on network QoS using some metrics such as Delay and Jitter. How to obtain these results? Is there already an integration with the FLow Monitor module?

DvdMgr, 2020-10-26:
Hi!
No integration with the FlowMonitor yet, I'm sorry. If you are interested
in implementing it, however, I would be happy to provide guidance.

A side note: if you are interested in measuring delay and jitter, you can
also do this without the FlowMonitor.

DvdMgr, 2020-11-17:
You can do this by tracking the packets in the network, and computing the time difference between when they are created by the ED application and when they are received by the GW - currently we provide no infrastructure to do this, but you can do it yourself if you connect the appropriate trace sources to keep track of the packets in your simulation script!

DvdMgr, 2021-03-22:
Closing because of inactivity.

closed 2021-03-22 by DvdMgr