signetlabdei/lorawan #95: question regarding InterferenceCallback

shobhitaggarwal92, 2020-10-25:
Hello,
I have a question regarding the InterferenceCallback.
The method gives back the number of packets interfered with on the gateway from all the devices.

I have created two device containers for 2 types of end devices (one for sf=7 and another for sf=8). I want to get just the number of interfered packets from sf=8 devices. Is this possible? 

Can you please help me or walk me through what function or which file to modify to get what I am looking for.

Thanks
Shobhit

DvdMgr, 2020-10-26:
Hi!
If you haven't modified it, the InterferenceCallback should actually return the gateway that is reporting the failure to receive the packet, not the number of packets.

I'm not sure I understand your question - are you interested in counting the number of packets using SF8 that suffered interference, or the number of packets that were lost because of interference from a packet using SF8?

shobhitaggarwal92, 2020-10-26:
> [*Caution*: Email from External Sender. Do not click or open links or
> attachments unless you know this sender.]

Thanks for the prompt reply.

For now I am interested in counting number of packets using sf=8 that are
lost due to interference.

In future I may need to count the the number of packets that were lost
because of interference from a packet using SF8.

I will really appreciate if you can walk me through both.

shobhitaggarwal92, 2020-10-26:
Thanks for the prompt reply.

For now I am interested in counting number of packets using sf=8 that are
lost due to interference.

In future I may need to count the the number of packets that were lost
because of interference from a packet using SF8.

I will really appreciate if you can walk me through both.

DvdMgr, 2020-11-17:
In this case, you can connect to the LoraPhy::LostPacketBecauseInterference trace source of the GW node: that way, your callback will be called every time a packet is lost because of interference, and you only need to check its Spreading Factor to see whether it's 8 or not.

DvdMgr, 2021-03-22:
Closing because of inactivity.

closed 2021-03-22 by DvdMgr