signetlabdei/lorawan #96: Enabling GUI visualisation on this module

charanpool, 2021-02-01:
## Expected Behavior
Able to visualize GUI plots on this module.

## Actual Behavior
No GUI plots are being produced besides showing the following error:
```
root@testuser-Vostro-3902:~/AtAI/Rack_Warehouse/LoRaSimulators/ns3_simulator/ns-3# ./waf --run complete-network-example --vis
Waf: Entering directory `/home/testuser/AtAI/Rack_Warehouse/LoRaSimulators/ns3_simulator/ns-3/build'
Waf: Leaving directory `/home/testuser/AtAI/Rack_Warehouse/LoRaSimulators/ns3_simulator/ns-3/build'
Build commands will be stored in build/compile_commands.json
'build' finished successfully (0.313s)
assert failed. cond="uid != 0", msg="Assert in TypeId::LookupByName: ns3::VisualSimulatorImpl not found", file=../src/core/model/type-id.cc, line=834
terminate called without an active exception
Command ['/home/testuser/AtAI/Rack_Warehouse/LoRaSimulators/ns3_simulator/ns-3/build/src/lorawan/examples/ns3-dev-complete-network-example-debug', '--SimulatorImplementationType=ns3::VisualSimulatorImpl'] terminated with signal SIGIOT. Run it under a debugger to get more information (./waf --run <program> --gdb").
```

## Steps to Reproduce the Problem

  1. Clone & install the repository as explained in readme.md
  2. Follow the steps to as mentioned on the official website(https://www.nsnam.org/wiki/PyViz) to enable GUI
  3. Build and Run as:
./waf configure --enable-tests --enable-examples
./waf build
./waf --run complete-network-example --vis

## Specifications

  - ns-3 version: same as provided
  - lorawan module version:same as provided
  - Platform:Ubuntu 18.04
  - Compiler:gcc version 7.5.0

DvdMgr, 2021-02-01:
It's expected visualization will not work with this module, since it is not supported. If you are interested in implementing this yourself, I can definitely assist - https://www.nsnam.org/wiki/PyViz would be a good resource to get started if you want to tackle this!

charanpool, 2021-02-01:
@DvdMgr , I have followed the exact same steps mentioned on the website under "Enabling the visualizer in your own simulation" section. However as mentioned above ended up with the error. Could you please help me fix it?

DvdMgr, 2021-02-01:
It seems like python bindings must be enabled to use pyviz, but python bindings are not currently supported by the simulator - did you build them yourself, too? Can you share the output of ./waf configure?

charanpool, 2021-02-01:
Yes, I built them too by installing required packages. 
```
root@testuser-Vostro-3902:~/AtAI/Rack_Warehouse/LoRaSimulators/ns3_simulator/ns-3# ./waf configure
Setting top to                           : /home/testuser/AtAI/Rack_Warehouse/LoRaSimulators/ns3_simulator/ns-3 
Setting out to                           : /home/testuser/AtAI/Rack_Warehouse/LoRaSimulators/ns3_simulator/ns-3/build 
Checking for 'gcc' (C compiler)          : /usr/bin/gcc 
Checking for cc version                  : 7.5.0 
Checking for 'g++' (C++ compiler)        : /usr/bin/g++ 
Checking for compilation flag -Wl,--soname=foo support : ok 
Checking for compilation flag -std=c++11 support       : ok 
Checking boost includes                                : headers not found, please provide a --boost-includes argument (see help) 
Checking boost includes                                : headers not found, please provide a --boost-includes argument (see help) 
Checking for program 'python'                          : /usr/bin/python3 
Checking for python version >= 2.3                     : 3.6.9 
python-config                                          : /usr/bin/python3-config 
Asking python-config for pyembed '--cflags --libs --ldflags --embed' flags : not found 
Asking python-config for pyembed '--cflags --libs --ldflags' flags         : yes 
Testing pyembed configuration                                              : yes 
Asking python-config for pyext '--cflags --libs --ldflags' flags           : yes 
Testing pyext configuration                                                : yes 
Checking for compilation flag -fvisibility=hidden support                  : ok 
Checking for compilation flag -Wno-array-bounds support                    : ok 
Checking for pybindgen location                                            : not found 
Checking for python module 'pybindgen'                                     : 0.21.0.post15+ga587377 
Checking for pybindgen version                                             : 0.21.0.post15+ga587377 
Checking for code snippet                                                  : yes 
Checking for types uint64_t and unsigned long equivalence                  : no 
Checking for code snippet                                                  : no 
Checking for types uint64_t and unsigned long long equivalence             : yes 
Checking for the apidefs that can be used for Python bindings              : gcc-LP64 
Checking for internal GCC cxxabi                                           : complete 
Checking for python module 'pygccxml'                                      : not found 
Checking for click location                                                : not found 
Checking for program 'pkg-config'                                          : /usr/bin/pkg-config 
Checking for gtk+-3 > 3.2                                                  : not found 
Checking for 'libxml-2.0'                                                  : not found 
checking for uint128_t                                                     : not found 
checking for __uint128_t                                                   : yes 
Checking high precision implementation                                     : 128-bit integer (default) 
Checking for header stdint.h                                               : yes 
Checking for header inttypes.h                                             : yes 
Checking for header sys/inttypes.h                                         : not found 
Checking for header sys/types.h                                            : yes 
Checking for header sys/stat.h                                             : yes 
Checking for header dirent.h                                               : yes 
Checking for header signal.h                                               : yes 
Checking for header pthread.h                                              : yes 
Checking for header stdint.h                                               : yes 
Checking for header inttypes.h                                             : yes 
Checking for header sys/inttypes.h                                         : not found 
Checking for library rt                                                    : yes 
Checking for header sys/ioctl.h                                            : yes 
Checking for header net/if.h                                               : yes 
Checking for header net/ethernet.h                                         : yes 
Checking for 'libdpdk'                                                     : not found 
Checking for header linux/if_tun.h                                         : yes 
Checking for header netpacket/packet.h                                     : yes 
Checking for header net/netmap_user.h                                      : not found 
Checking for NSC location                                                  : not found 
Checking for 'sqlite3'                                                     : not found 
Checking for header semaphore.h                                            : yes 
Checking for header linux/if_tun.h                                         : yes 
Checking for python module 'gi'                                            : 3.26.1 
Checking for python module 'gi.repository.GObject'                         : ok 
Checking for python module 'cairo'                                         : ok 
[... cut ...]
Real Time Simulator           : enabled
SQLite stats support          : not enabled (library 'sqlite3' and/or semaphore.h not found)
Tap Bridge                    : enabled
Tap FdNetDevice               : enabled
Tests                         : not enabled (based on configuration file)
Threading Primitives          : enabled
Use sudo to set suid bit      : not enabled (option --enable-sudo not selected)
XmlIo                         : not enabled (library 'libxml-2.0 >= 2.7' not found)
'configure' finished successfully (4.852s)
```

DvdMgr, 2021-02-01:
Ok, looks alright - did you also add `'visualizer'` as a module dependency for the `complete-network-example` program in `examples/wscript`?

charanpool, 2021-02-01:
Okay, Thanks for that. That fixed a part of the problem. Now I am able to see the plot of nodes and gateways.
[image]
But, the statistics of the node from the right-click options is not working. I see these log messages on terminal:
```
No OLSR
Traceback (most recent call last):
  File "/home/testuser/AtAI/Rack_Warehouse/LoRaSimulators/ns3_simulator/ns-3/src/visualizer/visualizer/plugins/ipv4_routing_table.py", line 147, in _show_ipv4_routing_table
    ShowIpv4RoutingTable(viz, node.node_index)
  File "/home/testuser/AtAI/Rack_Warehouse/LoRaSimulators/ns3_simulator/ns-3/src/visualizer/visualizer/plugins/ipv4_routing_table.py", line 81, in __init__
    self.visualizer.add_information_window(self)
  File "/home/testuser/AtAI/Rack_Warehouse/LoRaSimulators/ns3_simulator/ns-3/src/visualizer/visualizer/core.py", line 1697, in add_information_window
    info_win.update()
  File "/home/testuser/AtAI/Rack_Warehouse/LoRaSimulators/ns3_simulator/ns-3/src/visualizer/visualizer/plugins/ipv4_routing_table.py", line 103, in update
    routing = ipv4.GetRoutingProtocol()
AttributeError: 'NoneType' object has no attribute 'GetRoutingProtocol'
No OLSR
No OLSR
No OLSR
Traceback (most recent call last):
  File "/home/testuser/AtAI/Rack_Warehouse/LoRaSimulators/ns3_simulator/ns-3/src/visualizer/visualizer/plugins/ipv4_routing_table.py", line 147, in _show_ipv4_routing_table
    ShowIpv4RoutingTable(viz, node.node_index)
  File "/home/testuser/AtAI/Rack_Warehouse/LoRaSimulators/ns3_simulator/ns-3/src/visualizer/visualizer/plugins/ipv4_routing_table.py", line 81, in __init__
    self.visualizer.add_information_window(self)
  File "/home/testuser/AtAI/Rack_Warehouse/LoRaSimulators/ns3_simulator/ns-3/src/visualizer/visualizer/core.py", line 1697, in add_information_window
    info_win.update()
  File "/home/testuser/AtAI/Rack_Warehouse/LoRaSimulators/ns3_simulator/ns-3/src/visualizer/visualizer/plugins/ipv4_routing_table.py", line 103, in update
    routing = ipv4.GetRoutingProtocol()
AttributeError: 'NoneType' object has no attribute 'GetRoutingProtocol'
No OLSR
No OLSR
```
How do I fix it?

DvdMgr, 2021-02-01:
Well, that's a lot more than what I expected we were going to get. What we are up against now is the fact that the visualization module expects to find stuff that is not implemented in the lorawan module. For instance, in LoRaWAN networks you don't have an IP layer, but the visualizer here is looking for a routing protocol. I'm not sure how to proceed now, since I've never worked on or with the visualizer. I guess we could try and look at the visualizer documentation to see if it's possible to disable the fact that it looks for routing information. An alternative would be to create a dummy routing protocol to bypass the problem, but this solution is way less elegant.

charanpool, 2021-02-01:
One more thing to note is that, all the above changes are done after modifying to ```modules_enabled = ['all_modules']``` in file ".ns3rc". Making the ```modules_enabled = ['lorawan']``` will not link the _**Visualizer**_ specific code bundle to final build.

Thank you so much for helping me out. If I found some time, I will look into documentation and let us implement the complete backend GUI features in the module.

closed 2021-02-01 by charanpool