signetlabdei/lorawan #101: Default radius value causes devices to be out of range

JacoTuks, 2021-04-19:
## Expected Behavior
Setting the radius to 7500 m should allow all devices to still be able to communicate.

## Actual Behavior
This radius seems to be too large and when a simulation is run several devices are out of range. This can be seen by enabling line 534 of lorawan-mac-helper. A lower value such as 6300 m, does not have this problem. Perhaps a bug fix caused 7500 m to no longer be valid?

## Steps to Reproduce the Problem

  1. Enable line 534 of lorawan-mac-helper.cc NS_LOG_DEBUG ("Device out of range");
  2. Enable logging of this component in complete-network-example.cc (line 80)
  3. Run complete-network-example

## Specifications

  - ns-3 version: 3.30
  - lorawan module version:  Dev branch commit 5 March 2021 (2340df18f88132f2e8603ed2138e27429b7d2007)
  - Platform:  Ubuntu 20.04.2 LTS
  - Compiler:

DvdMgr, 2021-04-19:
A change in the propagation model parameters probably shrunk the coverage radius down. I don't think it's necessarily bad that the radius used in the example leaves some devices out of range however, since it's a situation that EDs can reasonably find themselves in, and it would be useful if the example made users aware of this. 

At the same time, I also agree that if a very large portion of devices is left out, we should reduce this number in the examples. Is this the case in the cited example?

JacoTuks, 2021-04-19:
For the default sim, 48 out of the 200 devices are out of range.

I agree that it's not a bad thing, but I do think it will be confusing when reading an older paper that state 7500 m allows all devices to be reachable and then running a sim which which does the opposite. Perhaps an additional comment in the files explaining the difference.

Using 6500 m causes 4/200 to be out of range.

6400 m and below allows all devices to be in range.

DvdMgr, 2021-04-19:
Didn't realize this was mentioned explicitly in the original paper! In this case, I think the best option is to lower the deployment radius to 6400 m as you mentioned. Feel free to open a pull request if you have the changes ready - I will do it myself otherwise.

JacoTuks, 2021-04-20:
I'm happy to send a pull request. I didn't do any sort of calculations for 6400, I just stepped it down in intervals of 100 until all devices could send. I assume you followed a similar approach for 7500?

DvdMgr, 2021-04-20:
Yes, it was trial and error as far as I can remember

DvdMgr, 2021-04-24:
Fixed in #102, closing.

closed 2021-04-24 by DvdMgr