signetlabdei/lorawan #107: Packet Delivery Ratio

k0nstant1na, 2021-05-24:
Hello
I am trying to measure the pdr (packet delivery ratio) by running adr-example. Is there a way to do that?

Thanks

renzoe, 2021-05-31:
Hi,  k0nstant1na !
(**this is nor really an "Issue**)
1st: what have you already tried?
 
I think for the adder-example, at you have two figures at the end of a simulation period , example 
`"4.000000 4.000000"`
this is packets sent; packet received

This is a pointer of what you are looking for!
have a good week!

Renzo

DvdMgr, 2021-06-01:
Hi, @renzoe is right! You can obtain PDR by computing received/sent.

On a side note: we probably need clearer / more examples and documentation, describing both how to compute metrics and how things work in general. If anyone is interested in tackling this, please contact us!

k0nstant1na, 2021-06-01:
Thank you. I did it as soon as I created the issue. Can i ask something else?  I don't know how to track air time. I know that the air transmission time is in lora-phy:Getonairtime, but I don't know how to add code in the adr-example to get the air transmission time of each device.

DvdMgr, 2021-06-03:
I don't think we currently have a trace source that exposes this information. You can add it yourself, following the example of `StartSending` in `LoraPhy`, for instance, and calling the trace source inside the `SimpleEndDeviceLoraPhy::Send` (in that function we also call `GetOnAirTime`, so you already have the Time object to pass to the trace source).

k0nstant1na, 2021-06-03:
Thank you so much.

closed 2021-06-03 by DvdMgr