signetlabdei/lorawan #108: Python bindings

xiaoyaodiwang, 2021-06-04:
Hi

Is Python bindings built for lorawan module? I'd like to call lorawan module using Python. Looking forward to your answer, thanks!

DvdMgr, 2021-06-07:
Hi, Python bindings are not bundled as of now. If I remember correctly, the last time I tried the command to create the bindings threw up an error, but it was quite some time ago. Let me know if you are successful in obtaining them ([this page](https://www.nsnam.org/docs/manual/html/python.html) should contain all needed instructions to generate them). I will try and look into this.

xiaoyaodiwang, 2021-06-07:
Hi, I've built bindings successfully using NS3 API Python scanning

DvdMgr, 2021-06-07:
Nice, good to know! I'll add them to the module once I get some time to work on this.

xiaoyaodiwang, 2021-06-23:
When I tried import lorawan module in Python, it reports that "ImportError: /ns-allinone-3.33/ns-3.33/build/bindings/python/ns/lorawan.cpython-36m-x86_64-linux-gnu.so: undefined symbol: _ZN3ns37lorawan39CorrelatedShadowingPropagationLossModel22GetCorrelationDistanceEv". How to solve this problem?

DvdMgr, 2021-06-24:
As a temporary workaround: can you try removing the `correlated-shadowing-propagation-loss-model.cc|h` files (also removing them from the `wscript`), provided you don't use them?

xiaoyaodiwang, 2021-06-25:
I've removed them, and then another error is reported: "lorawan.cpython-36m-x86_64-linux-gnu.so: undefined symbol: _ZN3ns37lorawan15EndDeviceStatus17UpdateGatewayDataESt3mapINS_7AddressENS1_15PacketInfoPerGwESt4lessIS3_ESaISt4pairIKS3_S4_EEES3_d". Then I tried to remove "end-device-status.cc|h" files, but it fails to compile without "end-device-status.h".

xiaoyaodiwang, 2021-10-10:
> Hi @xiaoyaodiwang, Did you succeed in binding the code to python?

Yes

DvdMgr, 2021-11-19:
Hi @xiaoyaodiwang, can you share how you were able to make this work? I'd like to include this feature in the mainline, if you want to!