signetlabdei/lorawan #116: How get the packet on NetworkServerHelper

xiaoyaodiwang, 2021-07-31:
I'm trying to get the packet received on NetworkServer and get the data content from the packet. However, there is no such methods in network-server.cc. Besides, the "StartApplication" and "StopApplication" method in network-server.cc file is empty so I don't know how to create the method that I want.

DvdMgr, 2021-08-04:
You can access the packets received from EDs via the `NetworkServer`'s `NetworkStatus` object. Check out the following modified version of `network-server-example.cc`, that prints the packet that was sent by the first ED.

```c++
/*
 * This example creates a simple network in which all LoRaWAN components are
 * simulated: End Devices, some Gateways and a Network Server.
 * Two end devices are already configured to send unconfirmed and confirmed messages respectively.
 */

#include "ns3/lora-device-address.h"
#include "ns3/lora-frame-header.h"
#include "ns3/point-to-point-module.h"
#include "ns3/forwarder-helper.h"
#include "ns3/network-server-helper.h"
#include "ns3/lora-channel.h"
#include "ns3/mobility-helper.h"
#include "ns3/lora-phy-helper.h"
#include "ns3/lorawan-mac-helper.h"
#include "ns3/lora-helper.h"
#include "ns3/gateway-lora-phy.h"
#include "ns3/periodic-sender.h"
#include "ns3/periodic-sender-helper.h"
#include "ns3/log.h"
#include "ns3/string.h"
#include "ns3/command-line.h"
#include "ns3/core-module.h"
#include "ns3/network-module.h"
#include "ns3/lora-device-address-generator.h"
#include "ns3/one-shot-sender-helper.h"

using namespace ns3;
using namespace lorawan;

NS_LOG_COMPONENT_DEFINE ("NetworkServerExample");

void
PrintReceivedPacketContents (Ptr<NetworkServer> ns, LoraDeviceAddress address)
{
  Ptr<Packet> packet = ns->GetNetworkStatus()->GetEndDeviceStatus(address)->GetLastPacketReceivedFromDevice()->Copy();
  NS_LOG_DEBUG ("Last packet received from device: " << packet);
  NS_LOG_DEBUG ("Packet contents: " << packet->ToString());
}

int
main (int argc, char *argv[])
{

  bool verbose = false;

  CommandLine cmd;
  cmd.AddValue ("verbose", "Whether to print output or not", verbose);
  cmd.Parse (argc, argv);

  // Logging
  //////////

  LogComponentEnable ("NetworkServerExample", LOG_LEVEL_ALL);
  LogComponentEnable ("NetworkServer", LOG_LEVEL_ALL);
  LogComponentEnable ("GatewayLorawanMac", LOG_LEVEL_ALL);
  // LogComponentEnable("LoraFrameHeader", LOG_LEVEL_ALL);
  // LogComponentEnable("LorawanMacHeader", LOG_LEVEL_ALL);
  // LogComponentEnable("MacCommand", LOG_LEVEL_ALL);
  // LogComponentEnable("GatewayLoraPhy", LOG_LEVEL_ALL);
  // LogComponentEnable("LoraPhy", LOG_LEVEL_ALL);
  // LogComponentEnable("LoraChannel", LOG_LEVEL_ALL);
  // LogComponentEnable("EndDeviceLoraPhy", LOG_LEVEL_ALL);
  // LogComponentEnable("LogicalLoraChannelHelper", LOG_LEVEL_ALL);
  LogComponentEnable ("EndDeviceLorawanMac", LOG_LEVEL_ALL);
  LogComponentEnable ("ClassAEndDeviceLorawanMac", LOG_LEVEL_ALL);
  // LogComponentEnable ("OneShotSender", LOG_LEVEL_ALL);
[... 108 lines cut ...]
  NetworkServerHelper networkServerHelper;
  networkServerHelper.SetGateways (gateways);
  networkServerHelper.SetEndDevices (endDevices);
  networkServerHelper.Install (networkServers);

  // Install the Forwarder application on the gateways
  ForwarderHelper forwarderHelper;
  forwarderHelper.Install (gateways);

  Packet::EnablePrinting();
  LoraDeviceAddress address = edLorawanMac1->GetDeviceAddress();
  Ptr<NetworkServer> ns = networkServers.Get(0)->GetApplication(0)->GetObject<NetworkServer>();
  Simulator::Schedule (Seconds(20), &PrintReceivedPacketContents, ns, address);

  // Start simulation
  Simulator::Stop (Seconds (800));
  Simulator::Run ();
  Simulator::Destroy ();

  return 0;
}

```

If you need to react to each packet you get at the NS, I suggest writing your custom `NetworkServerComponent`: all registered components (you need to modify `NetworkServerHelper::InstallComponents` to add your own) are notified whenever the NS receives a new packet. Check out the implementation of `ConfirmedMessagesComponent` or `LinkCheckComponent` to get an idea of how to implement it!

DvdMgr, 2021-11-19:
Closing due to inactivity.

closed 2021-11-19 by DvdMgr