signetlabdei/lorawan #119: How to send a package from the Network Server to the end devices

yacoubhanna, 2021-08-24:
Hello,

Is it possible to send a package from the Network Server to the end devices? If Yes, how can I do that?

I did run a complete network example, but the Network Server is only receiving packages from the end devices. I need to do the opposite of that. 

I would really appreciate it if you can help me with that. 

Thank you.

QiuYukang, 2021-08-25:
This model does not support this function, I have a rough implementation here, you can refer to it https://github.com/QiuYukang/lorawan/tree/qiu_dev

yacoubhanna, 2021-08-25:
Thank you for your response. Is your modified version has already implemented the issue that I'm facing or do I still have to write the code? Because what I'm trying to do is modify the network server example in order for the network server to send the package to the end devices. Is it possible to do that using your modified version?

asmanuha, 2021-08-30:
what a great jobs.. thanks @QiuYukang 
anyway I have tried my best to follow and modify my code as your suggest. However I have problem here

../src/lorawan/model/network-server.cc: In member function ‘void ns3::lorawan::NetworkServer::Send(ns3::Ptr<ns3::Packet>, ns3::lorawan::LoraDeviceAddress)’:
../src/lorawan/model/network-server.cc:192:16: error: ‘class ns3::lorawan::NetworkScheduler’ has no member named ‘DoSend’; did you mean ‘DoSet’?
   m_scheduler->DoSend(data, deviceAddress, 2);
                ^~~~~~
                DoSet
In file included from ./ns3/object.h:27:0,
                 from ./ns3/network-server.h:25,
                 from ../src/lorawan/model/network-server.cc:22:
./ns3/ptr.h: In instantiation of ‘ns3::Ptr<T>::Ptr(const ns3::Ptr<U>&) [with U = ns3::lorawan::EndDeviceLorawanMac; T = ns3::lorawan::ClassAEndDeviceLorawanMac]’:
../src/lorawan/model/network-server.cc:153:34:   required from here
./ns3/ptr.h:748:24: error: invalid conversion from ‘ns3::lorawan::EndDeviceLorawanMac*’ to ‘ns3::lorawan::ClassAEndDeviceLorawanMac*’ [-fpermissive]
   : m_ptr (PeekPointer (o))
            ~~~~~~~~~~~~^~~

could you give me advice, what 's wrong with my code?

asmanuha, 2021-08-30:
ooh... i found the answer about my first problem
just check your code in src/lorawan/modelnetwork-scheduler.h and add this line

/**
   * Send data to the end device in the specified recieve window
   */
  void DoSend(Ptr<Packet> data, LoraDeviceAddress deviceAddress, int window);

asmanuha, 2021-08-30:
still can't figure out what happen here

[1328/2134] Compiling src/lorawan/model/network-server.cc
In file included from ./ns3/object.h:27:0,
                 from ./ns3/network-server.h:25,
                 from ../src/lorawan/model/network-server.cc:22:
./ns3/ptr.h: In instantiation of ‘ns3::Ptr<T>::Ptr(const ns3::Ptr<U>&) [with U = ns3::lorawan::EndDeviceLorawanMac; T = ns3::lorawan::ClassAEndDeviceLorawanMac]’:
../src/lorawan/model/network-server.cc:153:34:   required from here
./ns3/ptr.h:748:24: error: invalid conversion from ‘ns3::lorawan::EndDeviceLorawanMac*’ to ‘ns3::lorawan::ClassAEndDeviceLorawanMac*’ [-fpermissive]
   : m_ptr (PeekPointer (o))
            ~~~~~~~~~~~~^~~

I really appreciate your help.. :)

and here also the problem that i found out: (I rename the class C example as classC under scratch) -- and I check the code here also put "GetMac instead of GetPhy" -- so where should i add the GetMac argument?

../scratch/classC.cc: In function ‘int main(int, char**)’:
../scratch/classC.cc:131:52: error: ‘class ns3::lorawan::LorawanMac’ has no member named ‘GetMac’; did you mean ‘GetPhy’?
   Ptr<EndDeviceLorawanMac> edLorawanMac1 = edMac1->GetMac ()->GetObject<EndDeviceLorawanMac> ();
                                                    ^~~~~~
                                                    GetPhy

many thanks

QiuYukang, 2021-08-31:
Hi @asmanuha ，are you going to get phy? 
Refer to this [example](https://github.com/QiuYukang/lorawan/blob/a8a6cf64de262a2d6af6d1b623807bbbc970e4d2/examples/class-c-end-device-example.cc#L131)，you can use：
```c++
Ptr<ClassCEndDeviceLorawanMac> edLorawanMac1 = edMac1->GetObject<ClassCEndDeviceLorawanMac> ();
Ptr<LoraPhy> edPhy = edLorawanMac1->GetPhy();
```
or
```c++
Ptr<ClassAEndDeviceLorawanMac> edLorawanMac1 = edMac1->GetObject<ClassAEndDeviceLorawanMac> ();
Ptr<LoraPhy> edPhy = edLorawanMac1->GetPhy();
```
Whether you use `ClassAEndDeviceLorawanMac` or `ClassCEndDeviceLorawanMac` depends on the type of end-device.

asmanuha, 2021-08-31:
> Hi @asmanuha if you want to send a packet from the Network Server to the end devices，you have to checkout to branch [qiu_dev](https://github.com/QiuYukang/lorawan).
> 
> Are you going to get phy?
> Refer to this [example](https://github.com/QiuYukang/lorawan/blob/a8a6cf64de262a2d6af6d1b623807bbbc970e4d2/examples/class-c-end-device-example.cc#L131)，you can use：
> 
> ```c++
> Ptr<ClassCEndDeviceLorawanMac> edLorawanMac1 = edMac1->GetObject<ClassCEndDeviceLorawanMac> ();
> Ptr<LoraPhy> edPhy = edLorawanMac1->GetPhy();
> ```
> 
> or
> 
> ```c++
> Ptr<ClassAEndDeviceLorawanMac> edLorawanMac1 = edMac1->GetObject<ClassAEndDeviceLorawanMac> ();
> Ptr<LoraPhy> edPhy = edLorawanMac1->GetPhy();
> ```
> 
> Whether you use `ClassAEndDeviceLorawanMac` or `ClassCEndDeviceLorawanMac` depends on the type of end-device.

hi @QiuYukang thanks for your answer, actually I want to "GetMac" so that I can see information in the Layer 2

QiuYukang, 2021-08-31:
hi @asmanuha ，can you upload the script file which you has modified, so that I can help you troubleshoot the problem better?

asmanuha, 2021-08-31:
hi **@QiuYukang**  Many thanks for your fast response,
I have step by step as here

1. Create file "lorawan/model/class-c-end-device-lorawan-mac.h" copy paste from Reference -->  https://github.com/QiuYukang/lorawan/blob/qiu_dev/model/class-c-end-device-lorawan-mac.h 

2. Create file "lorawan/model/class-c-end-device-lorawan-mac.cc" copy paste from Reference -->  https://github.com/QiuYukang/lorawan/blob/qiu_dev/model/class-c-end-device-lorawan-mac.cc

3. Add a line of code in "lorawan/model/lora-device-address.h"  Reference -->  https://github.com/QiuYukang/lorawan/blob/a8a6cf64de262a2d6af6d1b623807bbbc970e4d2/model/lora-device-address.h
> bool IsBroadcast ();

4. Add 5 line of code in file "lorawan/model/lora-device-address.cc" around line no 165
bool 
LoraDeviceAddress::IsBroadcast ()
{
  return m_nwkId.Get () == 0x7F && m_nwkAddr.Get () == 0x1FFFFFF;
}
Reference --> https://github.com/QiuYukang/lorawan/blob/a8a6cf64de262a2d6af6d1b623807bbbc970e4d2/model/lora-device-address.cc#L166

5.Add a line of code in file "lorawan/model/network-server.h"
 void Send(Ptr<Packet> data, LoraDeviceAddress deviceAddress);
Refrence --> https://github.com/QiuYukang/lorawan/blob/01816f45623ba964738d062216fc1830d9f3b362/model/network-server.h

6. Add line of codes in File "lorawan/model/network-server.cc" 

== around  line number 184
void NetworkServer::Send(Ptr<Packet> data, LoraDeviceAddress deviceAddress)
{
  NS_LOG_FUNCTION(this << data << deviceAddress);

  m_scheduler->DoSend(data, deviceAddress, 2);
}

== around line 144
// Get the MAC
 // Ptr<ClassAEndDeviceLorawanMac> edLorawanMac =
 //   loraNetDevice->GetMac ()->GetObject<ClassAEndDeviceLorawanMac> ();
 // Ptr<ClassCEndDeviceLorawanMac> edLorawanMac1 =
 //   loraNetDevice->GetMac ()->GetObject<ClassCEndDeviceLorawanMac> ();
  Ptr<EndDeviceLorawanMac> edLorawanMac =
      loraNetDevice->GetMac ()->GetObject<EndDeviceLorawanMac> ();

  // Update the NetworkStatus about the existence of this node
  m_status->AddNode (edLorawanMac);

===
7. disable/ make it as command -the "private"- in file lorawan/model/end-device-lorawan-mac.h (around line 434) 
//private:     
Reference --> https://github.com/QiuYukang/lorawan/blob/01816f45623ba964738d062216fc1830d9f3b362/model/end-device-lorawan-mac.h

8. Add line of codes in file "lorawan/model/end-device-lorawan-mac.cc"
===== around line number 155 up to 162

  // Check that payload length is below the allowed maximum
  if (packet->GetSize () > m_maxAppPayloadForDataRate.at (m_dataRate))
    {
      NS_LOG_WARN ("Attempting to send a packet larger than the maximum allowed"
                   << " size at this DataRate (DR" << unsigned(m_dataRate) <<
                   "). Transmission canceled.");
      return;
    }

=====Also add a line of code around line number 285
 m_currentFCnt++;

Reference --> https://github.com/QiuYukang/lorawan/blob/qiu_dev/model/end-device-lorawan-mac.cc

9. Add a line of codes in File "lorawan/model/network-scheduler.h"
== line number  
  void DoSend(Ptr<Packet> data, LoraDeviceAddress deviceAddress, int window);

[... 152 lines cut ...]
      else if (classCMac != 0)
        {
          tag.SetDataRate (classCMac->GetSecondReceiveWindowDataRate ());
        }
      tag.SetFrequency (edStatus->GetSecondReceiveWindowFrequency ());
      break;
    }

  packet->AddPacketTag (tag);
  return packet;
}

==== == make changes in 235 up to 246
Ptr<Packet>
NetworkStatus::GetReplyForDevice (LoraDeviceAddress edAddress, int windowNumber)
{
  // Get the reply packet
  Ptr<EndDeviceStatus> edStatus = m_endDeviceStatuses.find (edAddress)->second;
  Ptr<Packet> packet = edStatus->GetCompleteReplyPacket ();
  Ptr<ClassAEndDeviceLorawanMac> classAMac =
      edStatus->GetMac ()->GetObject<ClassAEndDeviceLorawanMac> ();
  Ptr<ClassCEndDeviceLorawanMac> classCMac =
      edStatus->GetMac ()->GetObject<ClassCEndDeviceLorawanMac> ();
  NS_LOG_DEBUG ("Class_A_Mac:" << classAMac << "  Class_C_Mac:" << classCMac);
  NS_ASSERT (classAMac != 0 || classCMac != 0);

QiuYukang, 2021-08-31:
Hi @asmanuha, did you make these changes based on the [latest branch](https://github.com/signetlabdei/lorawan/tree/9f373c23076c3e821e6a6104bde8409e1a24a9fe) of `signetlabdei/lorawan`? 
I don’t think my changes will work well on the latest branch。
In fact, I made these changes based on [this commit](https://github.com/signetlabdei/lorawan/tree/474402318be571a6eda4fc214b22ebe5908942ed), you can refer to it：
https://github.com/QiuYukang/lorawan/compare/474402318be571a6eda4fc214b22ebe5908942ed...a8a6cf64de262a2d6af6d1b623807bbbc970e4d2

asmanuha, 2021-08-31:
Hi **@QiuYukang** Yes, I did make the changes from the latest version.
So to make it easier should I just need to remove folder (reinstall) the "lorawan" in the ns3 and changes the lorawan based on this reference : https://github.com/signetlabdei/lorawan/tree/474402318be571a6eda4fc214b22ebe5908942ed ?

Then after the Network server mange to send the packet to EndDevice, I can upgrade the lorawan version.
is it right?

My final goal is : Network server can send the "a piece of security information" to End Device, and the end device can follow up that information.

QiuYukang, 2021-08-31:
Hi @asmanuha, I have merge my branch with the latest commit of `signetlabdei/lorawan/develop`, the latest version is in [`QiuYukang/lorawan/class-c`](https://github.com/QiuYukang/lorawan).

I think you can do you work on [`QiuYukang/lorawan/class-c`](https://github.com/QiuYukang/lorawan) to send the "a piece of security information" to End Device.

asmanuha, 2021-08-31:
Hi **@QiuYukang** many thanks for your help .. I really appreciate your hard work.
OK, I will try now

asmanuha, 2021-09-01:
Hi **@QiuYukang** . It Works !!!
Thank you so much ツ
it's good work...

yacoubhanna, 2021-09-20:
Hello @asmanuha , I'm facing the same problem now. In which I'm trying to send a packet or message from the network server to the end device and getting the same error as yours. Did you have to modify a lot of stuff or just add some codes? I would really appreciate it if you can share what you did in order to work. 

Thank you.

asmanuha, 2021-09-20:
Hi, @yacoubhanna. Yes,  I modified 13 files according to the information. I recorded every modification, and the total is around 600 line codes. I think it is not polite to add a very long message in this command section. I am still a newbie playing with Github, and I don't know where the best way to put the modification file is (which the code itself is not mine); so that everyone can easily use it.

yacoubhanna, 2021-09-20:
Hello @asmanuha, thanks for your response. I think what you can do is to upload the helper and the model folders with the code example to your Github and place only the modified files there the .h and .cc files, and you can mention that these are modified files from the LoRaWan source, and I can modify the files from my end. If it is possible. @QiuYukang since you are one of the contributors is it okay to do this?  

Thank you.

yacoubhanna, 2021-09-21:
Hello @asmanuha. I believe there will be no problem posting the modified files on your Github or maybe send them by email. Please let me know if you can help me with that. Since I'm still having some trouble with it. 

Thank you.

asmanuha, 2021-10-02:
Hi @QiuYukang, many thanks for your help, and since @yacoubhanna gave me a question in my Github, I put the configuration's change there. Here is the step by step link: https://github.com/asmanuha/LoRaWAN/issues/1#issuecomment-932696356 

*sorry for my late response

yacoubhanna, 2021-10-03:
Hello, @asmanuha Thank you so much.

DvdMgr, 2021-11-19:
It looks like this was solved. Thanks to everyone that got involved! 

We have considered adding this as a feature in the past, and probably will at some point in the future - unfortunately, other commitments make it a bit hard as of now to follow all developments and actively work on new feature. 

Closing for now.

closed 2021-11-19 by DvdMgr

mouaadh1010, 2022-10-25:
Hi everyone ,
in lorawan's network-server-example, I cannot define, configure and interpret the ED address, GW address and NewServ address, if you can help me I will be very [grateful
[image]