signetlabdei/lorawan #120: how to set packet loss rate in lorawan module

xiaoyaodiwang, 2021-09-22:
I'd like to explore what may happen in the case of packet loss, so I need to set PHY packet loss rate between two nodes. How to set packet loss rate between two nodes in lorawan module?

CodeZombie, 2021-11-05:
Usually packet loss is a measurement of how many packets were successfully received vs how many were lost due to interference/collisions. You don't control packet loss directly, its just a result of many factors in the network. 

It seems however that you want to set up the network in such a way that packets will intentionally be lost at a given rate. There's a number of ways you could do this, but they're all going to have different characteristics and side effects.

One simple way would be to modify the SimpleGateWayLoraPhy::EndReceive method to randomly pretend there was interference at a given frequency, and destroy uplink packets before they're received. This, however, still wont give you an exact packet loss rate, as other factors could still cause interference/packet failure.

DvdMgr, 2021-11-19:
I think @CodeZombie's reply is quite good, thanks for taking the time to address this question!

closed 2021-11-19 by DvdMgr