signetlabdei/lorawan #121: How to simulate LoRaWAN Backend: Join, Network & Application Server?

asmanuha, 2021-10-07:
## Expected Behavior to simulate authentication and encryption
Hi, I want to simulate the LoRaWAN Backend system, in which the system consists of Join, Network, and Application servers. The use of this system is to simulate security features owned by the LoRaWAN. **Could you give me a hint on how to start to build them?**

## Actual LoRaWAN Behavior according to standardisation
- Node (End Device) send Join Request message to Join Server (via Gateway and Network Server)
- Join Server reply the message by sending Join Accept to End Device (forwarded via Network Server)
- Both Eend Device and Join Server executed Key Derivation Function Locally --> the results are AppSKey and NwkSKey(s)
- Join Server send the AppSKey to Application Server, and send the NwkSKey(s) to Network Server
- The AppSkey used as session key to encrypt payload between End Device and Application Server
- The NwkSKey(s) used as session key to encrypt payload between End Device and Network Server

Note: To simplify the explanation I write NwkSKey(s) to represent NwkSIntKey, SNwkSIntKey and NwkSEncKey

## Specifications
  - lorawan module version: 1.1

DvdMgr, 2021-11-19:
Hi!

Regarding the join procedure message exchanges, I think this would fit naturally either as a new `NetworkControllerComponent` or as additional code in `NetworkServer`.

I noticed that you also mention encrypting the payloads: typically, system level simulators such as this one do not model encryption in full detail, since they are more concerned about correctly modeling packet exchanges. Actually encrypting the data slows down simulations for no additional insight, so I don't see any advantage in modeling encryption, too.

asmanuha, 2021-11-19:
Hi, @DvdMgr thanks for your response. Could you give me a suggestion? 

**What kind of simulator can we use to simulate the full package of security features within LoRaWAN's environment?** My research is about LoRaWAN security, and I should simulate a certain security algorithm before directly implementing it in the hardware. I hope that doing the simulation can reduce the potential problem of the hardware itself. Actually, hardware implementation is beyond the scope of my research. I need to prove that my proposed security algorithm is proven in the end-to-end LoRaWAN topology by simulating it.

Or is there anyone who has the experience so that you can share it with me? I'll really appreciate it. 

Thanks

DvdMgr, 2021-11-20:
I'm not sure about what the appropriate tools to do security research would be. Depending on your needs, this module might be good enough but this is not my area of expertise, sorry.

I'll leave this issue open for some time to let other people who might help chime in!