signetlabdei/lorawan #129: bash: ./waf: No such file or directory

jndqadar, 2022-11-13:
Hi, 

I am compiling this repository, however, it gives me the following error given below. Can someone please help me out?
bash: ./waf: No such file or directory

[image]

sanjay-charran, 2022-11-15:
The "waf" script was replaced by the "ns3" one in the newer versions of ns3 I believe. This command ought to work for configuring for your version:
`./ns3 configure --enable-tests --enable-examples`

However, this module was designed for ns3.29 according to the "NS3-VERSION" file in the repository and I'm not certain if it'll work, so I would recommend you try to install ns3.29 if compiling the repository on the version you're running fails.

jndqadar, 2022-11-15:
@sanjay-charran Thank you so much, you are right. I then tried with `./ns3 configure --enable-tests --enable-examples` and compiled successfully. 
Thanks again

usamamustafa36, 2023-09-01:
I have replaced the waf script with ns3. it worked. however, when you run the test, it doesnt work and this error is shown. Can someone help ?
[image]

usamamustafa36, 2023-09-01:
The test was passed but when i run the example this error comes.. i am using ./ns3 instead of waf script
[image]

sanjay-charran, 2023-09-01:
@usamamustafa36 I believe it should be `./ns3 run [example-name]`

https://www.nsnam.org/docs/tutorial/html/quick-start.html

pagmatt, 2023-10-25:
Updated the README to reflect the build system changes.

closed 2023-10-25 by pagmatt