signetlabdei/lorawan #149: test.py failing with "Packet changed contents when going through the channel"

Ferrari3rddriver, 2023-12-24:
Hello guys

## Expected Behavior

Finally, make sure tests run smoothly with:

./test.py -s lorawan

Expectation is to pass all tests.

## Actual Behavior

$ ./test.py -s lorawan --text=loratest
Finished executing the following commands:
/usr/bin/cmake --build /home/jurca/Desktop/USA/PriviNet/Simulation-Demo/ns-3-dev-git/cmake-cache -j 1
[0/1] FAIL: TestSuite lorawan
0 of 1 tests passed (0 passed, 0 skipped, 1 failed, 0 crashed, 0 valgrind errors)
List of FAILed tests:
    lorawan

Writing results to text file "loratest.txt"...done.

$ cat loratest.txt 
FAIL: Test Suite "lorawan" (0.005)
PASS: Test Case "Verify that LoraInterferenceHelper works as expected" (0.000)
PASS: Test Case "Verify that LoraDeviceAddress works as expected" (0.000)
PASS: Test Case "Verify that LorawanMacHeader and LoraFrameHeader work as expected" (0.001)
PASS: Test Case "Verify that ReceivePaths work as expected" (0.000)
PASS: Test Case "Verify that LogicalLoraChannel and LogicalLoraChannelHelper work as expected" (0.000)
PASS: Test Case "Verify that LoraPhy"s function to compute the time on air of a packet works as expected" (0.000)
FAIL: Test Case "Verify that PhyConnectivity works as expected" (0.003)
    Details:
      Message:   Packet changed contents when going through the channel
      Condition: HaveSamePacketContents(packet, m_latestReceivedPacket) (actual) == true (limit)
      Actual:    0
      Limit:     1
      File:      /home/jurca/Desktop/USA/PriviNet/Simulation-Demo/ns-3-dev-git/src/lorawan/test/lorawan-test-suite.cc
      Line:      1481

## Steps to Reproduce the Problem

  1. Install ns-3 according to README.md instruction from https://github.com/nsnam/ns-3-dev-git. Run tests after installing is 
complete.
  2. Install lorawan module according to README.md instruction from https://github.com/signetlabdei/lorawan/tree/develop. Run tests for lorawan after installing is complete.
  

## Specifications

  - ns-3 version:
  ns-3 dev version from here: https://github.com/nsnam/ns-3-dev-git
  Repo was cloned and built on 24 December 2024.
  Master branch was used:
  $ git branch
  * master
  Output of VERSION file:
  $ cat VERSION 
  3-dev

  - lorawan module version:
  lorawan version from here: https://github.com/signetlabdei/lorawan/tree/develop
  Repo was cloned and built on 24 December 2024.
  Develop branch was used:
  $ git branch
  * develop
  Output of VERSION file:
  $ cat VERSION 
  0.3.1

  - Platform:
  Ubuntu Virtual Machine available in Oracle VirtualBox.
  $ cat /etc/os-release
  NAME="Ubuntu"
  VERSION="20.04.6 LTS (Focal Fossa)"
  ID=ubuntu
  ID_LIKE=debian
  PRETTY_NAME="Ubuntu 20.04.6 LTS"
  VERSION_ID="20.04"
  HOME_URL="https://www.ubuntu.com/"
  SUPPORT_URL="https://help.ubuntu.com/"
  BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
  PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
  VERSION_CODENAME=focal
  UBUNTU_CODENAME=focal

  - Compiler:
  Cmake was uesd.
  $ cmake --version
  cmake version 3.16.3

  CMake suite maintained and supported by Kitware (kitware.com/cmake).

non-det-alle, 2024-01-19:
Hello,

Thank you for the issue and sorry for the late reply.

Your problem may be related to using a more recent ns-3 version than the one supported, or by the installation guide in the readme being a bit out of date. I committed an updated version of the readme with better explanations and more automation in scripts, could you please try it out?

Ferrari3rddriver, 2024-01-27:
Thank you.
I had the chance to try on different setup, newer setup in terms of version.
[0/1] PASS: TestSuite lorawan
1 of 1 tests passed (1 passed, 0 skipped, 0 failed, 0 crashed, 0 valgrind errors)

I guess it had to do with my older virtual machine setup, not critical anyhow I would say, even though the test failed simulation could be ran.

Ferrari3rddriver, 2024-01-27:
Your answer is much appreciated.

closed 2024-01-27 by Ferrari3rddriver