signetlabdei/lorawan #150: m_receiveCallback in Receive function in lora-net-device

chekraF, 2024-02-09:
This instruction generates an error: msg="Attempted to dereference zero pointer". I try to find the problem, and I want to ask why you filled empty empty stuff in ptotocol and address m_receiveCallback (this, packet, 0, Address()); . the callback should not contain th eaddress of the gateway as aparameter?

non-det-alle, 2024-02-09:
> This instruction generates an error: msg="Attempted to dereference zero pointer".

This is probably because you are calling `m_receiveCallback` (i.e., `LoraNetDevice::Receive`) before setting the callback with `LoraNetDevice::SetReceiveCallback`. The pointer to the callback function is still zero and so the error message. This way of forwarding packets up the stack is normal practice in ns-3 (compare for instance ns-3's WiFi module).

> you filled empty empty stuff in protocol and address m_receiveCallback (this, packet, 0, Address()); .

This is because - in the real world -  the LoRaWAN stack (PHY + MAC layers) does not have an actual NetDevice in the traditional network sense. Unfortunately, ns-3 expects you to always have a NetDevice (it is part of the base ns-3 abstractions). As a consequence, in the current implementation packets are just forwarded through the NetDevice object, going from the MAC layer to the Application. We actually have a future proposal to completely remove the NetDevice from the LoRaWAN stack for this reason.

> The callback should not contain the address of the gateway as a parameter?

LoRaWAN gateways do not have a LoRaWAN address (i.e., a DevAddr) so there is nothing to put there on the LoRaWAN stack side. However, they usually do have an IP for forwarding packets from the LoRaWAN side to a server. This is reflected in the simulation, as they can have a second NetDevice (CSMA for instance) which can store their address and have an implemented version of the callback you mention.

chekraF, 2024-02-09:
The SetReceiveCallback( ...) is already called in the network-server-helper, then at gateway-lorawan-mac-level in GatewayLorawanMac::Receive  the Receive( ) function in net-device is called where we find the callback!  All these implementation already exist and I didn't modify it ...however it still stopping at lora-net-device. level exactly at m_receiveCallback to generate msg="Attempted to dereference zero pointer error!

non-det-alle, 2024-02-11:
Are you sure the callback in `GatewayLorawanMac::Receive` is the one giving you problems? Gateways' callbacks are set in [`forwarder-helper`](https://github.com/signetlabdei/lorawan/blob/be7884b18bc6032b7cb7cb19369b68504ee6e2b1/helper/forwarder-helper.cc#L88), not in `network-server-helper`. The callback that gives you problems may be the one set on the gateway's `PointToPointNetDevice`.

I think I understand where your error is coming from. In your example main file, make sure that `NetworkServerHelper::Install` is called before `ForwarderHelper::Install`. This is because the `NetworkServerHelper` installs a P2P NetDevice on the gateways, which is then needed by `ForwarderHelper` to set a receive callback for messages going from the server to the gateway through the P2P link. If this condition is not respected, the `ForwarderHelper` will skip installing the callback and thus your error message. 

Ideally this should give a better error message during installation that explains the issue. This is a known problem of the module, and it is already scheduled to be fixed in upcoming releases. Sorry for the inconvenience.

chekraF, 2024-02-12:
Exactly! Thank you so much for the clarification!! Your indication has resolved the problem! I kindly ask you to keep this kind of remarks in a short tuto with the installation setup! it would be very useful! I was dealing with this error for a whole week!
Thank you again for the instructions.

closed 2024-03-20 by non-det-alle