signetlabdei/lorawan #152: Performance metrics

Benoflynn, 2024-02-15:
Hi I've been using your complete LoRaWAN network example and was first of all wondering what performance metrics are being captured when running? From what i can tell the 200 refers to the number of nodes but im not sure about the 196. 
I also wanted to make some adjustments to measure [packet loss rates, throughput and delays Vs. The number of nodes] with this example and was wondering was there an easy way to do this.
Thank You!

  - ns-3 version: 3.36.1
  - lorawan module version:
  - Platform:
  - Compiler:

non-det-alle, 2024-02-15:
Hello, the 200 is the total number of packets sent by the LoRaWAN devices, while the 196 is the number of received packets by the network server. [This](https://github.com/signetlabdei/lorawan/blob/be7884b18bc6032b7cb7cb19369b68504ee6e2b1/examples/complete-network-example.cc#L332) is the line in `complete-network-example.cc` responsible for printing the values on screen. All metrics and metrics-related functions can be found in the `LoraPacketTracker` class, for instance [this](https://github.com/signetlabdei/lorawan/blob/be7884b18bc6032b7cb7cb19369b68504ee6e2b1/helper/lora-packet-tracker.cc#L340) is the line providing the answer to your question about numbers printed on screen. You could edit the `LoraPacketTracker` class to compute the metrics you are interested in and then print it in a file as it is done [here](https://github.com/signetlabdei/lorawan/blob/be7884b18bc6032b7cb7cb19369b68504ee6e2b1/examples/adr-example.cc#L277C1-L283C96). These are just some pointers on how to proceed, I suggest you study the code and the ns-3 documentation in order to extract from the simulation objects all the data you need. Make sure you are using the latest version of the code (you wrote that you are on ns-3.36.1, while currently the module supports ns-3.40)!

Benoflynn, 2024-02-15:
Thanks so much, I appreciate the quick and detailed feedback

closed 2024-02-15 by non-det-alle